rosjava/rosjava_core #203: Release of .deb packages for Indigo

knowrob, 2015-01-23:
While the current installation procedure works, it is rather complicated and makes it hard to distribute code based on rosjava. In addition, compilation is slower if several core packages have to be installed from source.

Is a binary release for Indigo planned? What are the blockers that need to be fixed?

pcernocky, 2015-02-12:
I would be hapy for debs too.
How are these deb pacakges built? Is somewhere a script to build them, maybe even for hydro?
If someone directs me a bit, I can help with it.

stonier, 2015-02-20:
Had to work through some changes with the messaging system before getting to the debs. Hydro was awkward keeping them up to date.

Almost done now though - would be happy to teach one of you how to maintain them once I've got a first release out. I won't be working with java over the next year.

stonier, 2015-02-20:
@pcernocky do you want to tee up a hangout to discuss sometime?

stonier, 2017-04-03:
Now on to kinetic....and Julian has the ball :)

closed 2017-04-03 by stonier