rosjava/rosjava_core #205: not all the tests are pass.

talregev, 2015-02-18:
@stonier 
I run the tests, and not all of them are passing.

I will try to investigate. it very important that all tests will be pass, and then we can continue to develop, 
to see if we don't break anythings. 

Tal

stonier, 2015-02-20:
I don't get any failure on the first...but on the ntp test, I think the problem deeper. Doesn't matter even if I set it out to 12s and use a local ntp server, the ntp library's `getTime` call just blocks indefinitely from time to time. e.g. a result of a failed run:

```
  <system-out><![CDATA[New Ros Core on : http://127.0.0.1:57088/
Registering pubisher [/ntp_time_provider][/rosout][rosgraph_msgs/Log]
onStart [1424440242058]
updateTime [1424440242058]
Updating time offset from NTP server: asia.pool.ntp.org [1424440242058]
Got time [1424440242469]
Updating time offset from NTP server: asia.pool.ntp.org [1424440242469]
Got time [1424440242874]
Updating time offset from NTP server: asia.pool.ntp.org [1424440242874]
Latch waiting : 1 false [1424440253892]
]]></system-out>
```

stonier, 2015-02-20:
The ntpclient was configured to block indefinitely by default. I set a timeout on it to 500ms instead and added some code to check for too many failures as it looped. Now seems to work as intended:

```
  <system-out><![CDATA[New Ros Core on : http://127.0.0.1:35824/
Registering pubisher [/ntp_time_provider][/rosout][rosgraph_msgs/Log]
onStart [1424441770828]
updateTime [1424441770828]
Updating time offset from NTP server: asia.pool.ntp.org [1424441770828]
Updating time offset from NTP server: asia.pool.ntp.org [1424441771219]
Updating time offset from NTP server: asia.pool.ntp.org [1424441771604]
Updating time offset from NTP server: asia.pool.ntp.org [1424441771990]
Updating time offset from NTP server: asia.pool.ntp.org [1424441772374]
Updating time offset from NTP server: asia.pool.ntp.org [1424441772759]
Updating time offset from NTP server: asia.pool.ntp.org [1424441773144]
Failed to read time from NTP server: asia.pool.ntp.orgReceive timed out
Updating time offset from NTP server: asia.pool.ntp.org [1424441773647]
Updating time offset from NTP server: asia.pool.ntp.org [1424441774032]
Updating time offset from NTP server: asia.pool.ntp.org [1424441774418]
Updating time offset from NTP server: asia.pool.ntp.org [1424441774804]
NTP time offset: 23 ms
System time: 1424441775190
NTP time: 1424441775213
Latch count: 1
Latch down : 0
Latch waiting : 0 true [1424441775191]
]]></system-out>
```

stonier, 2015-02-20:
Renamed this issue to focus on the ntp test.

@talregev open a new issue for the other one with more data if you are still having the problem.

closed 2015-02-20 by stonier