rosjava/rosjava_core #208: Create anonymous node in rosjava

talregev, 2015-03-12:
Here a link to simple ros node: [listner](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29)

there is a way to create anonymous node:
'''
    # In ROS, nodes are uniquely named. If two nodes with the same
    # node are launched, the previous one is kicked off. The
    # anonymous=True flag means that rospy will choose a unique
    # name for our 'listener' node so that multiple listeners can
    # run simultaneously.
rospy.init_node('listener', anonymous=True)
'''
How i can do the same functionality in rosjava?

Thanks,
Tal.

stonier, 2015-03-13:
I haven't done it before, but I assume it would be when you write a class that extends `AbstractNodeMain`.

When you override `getDefaultNodeName`, instead of returning `Graphname` as some string, then use `GraphName.newAnonymous()` call. That'll just give you a prefix though, so you might want to concatenate it.

Code is in https://github.com/rosjava/rosjava_core/blob/indigo/rosjava/src/main/java/org/ros/namespace/GraphName.java

stonier, 2015-03-13:
Better place for questions is [ros answers](http://answers.ros.org/).

talregev, 2015-03-13:
i will :+1:

jubeira, 2017-11-07:
@talregev did the answer above work for you? can we close this one?

This could be somewhere in a tutorial, the question was indeed interesting.

talregev, 2017-11-07:
Yes, you can close.

closed 2017-11-07 by jubeira