rosjava/rosjava_core #209: Implement cv_bridge on rosjava

talregev, 2015-03-21:
can it Implement cv_bridge for rosjava?
https://github.com/ros-perception/vision_opencv

Tal.

talregev, 2015-08-16:
I implement most of the functionality of [android cv_bridge][1].
It android because OpenCV is for Android and not java alo ne.
But some of the functionality is missing for me.
there is equivalent methods and constant from ros c++ in rosjava?  
#include <sensor_msgs/image_encodings.h>
sensor_msgs::image_encodings::isColor()
sensor_msgs::image_encodings::isMono()
sensor_msgs::image_encodings::isBayer()
sensor_msgs::image_encodings::numChannels()
sensor_msgs::image_encodings::bitDepth()
sensor_msgs::image_encodings::RGB8     (constant)
sensor_msgs::image_encodings::YUV422  (constant)

Can you help?
@stonier 

[1]: https://github.com/talregev/android_cv_bridge

kejriwalnishant1990, 2015-08-18:
Can you provide a sample code for usage of this library?

talregev, 2015-08-18:
From not completed [Wiki:][1]
```Java
//Converting ROS image messages to OpenCV images
CvImage cv_image = CvImage.toCvCopy(image_message, "bgr8");

//Converting OpenCV images to ROS image messages
//you need to create your own node and your own publisher, and a new empty msgs.
resolver = node.getResolver().newChild("/camera");
rawImagePublisher = node.newPublisher(resolver.resolve("image/raw"), Image._TYPE);
Image rawImage = rawImagePublisher.newMessage();

//code using the library:
CvImage cvImage = new CvImage(cv_image, "bgr8");
image_message = cvImage.toImageMsg(rawImage);
```

[1]: http://wiki.ros.org/cv_bridge/Tutorials/Converting%20between%20ROS%20images%20and%20OpenCV%20images%20%28Android%20Java%29#preview

kejriwalnishant1990, 2015-08-18:
Suppose if I have image_message(sensor_msgs/Image) in the callback function, then i can convert image_message to cv::Mat using following lines:
CvImage cv_image = CvImage.toCvCopy(image_message, "bgr8");

So i will get image in as a opencv image type (CvImage) and i can convert it into cv::Mat.
Is it right?

talregev, 2015-08-18:
in `CvImage cv_image` you have a public memer call image.
`cv_image.image` is a cv::Mat.

talregev, 2015-08-18:
I you want to help, I will appreciate if you update some of the Wiki, and write a small example of your own. thanks! :)

jubeira, 2017-05-29:
(See https://github.com/rosjava/rosjava_extras/pull/16)

closed 2017-05-29 by talregev