rosjava/rosjava_core #233: ListenerGroup.shutdown() never called

stratomda, 2017-02-27:
The ListenerGroup.shutdown() method is never called when cleaning up the DefaultNode class. I believe that this is causing a memory leak issue with android_core when trying to shutdown objects that extend NodeMain. See: http://answers.ros.org/question/255722/rosjava-listenergroup-causes-android-memory-leak/

adamantivm, 2017-03-01:
Thanks for the report @stratomda . I'll try to dig into it when I can get a chance. If you get to it before I do, by all means please submit a PR.

Meanwhile, I'll leave the link to your test code [here](https://github.com/stratomda/ROSLeak).

stratomda, 2017-03-02:
@adamantivm I think I have a fix for this. I have started testing it and should be done in the next few days. This same issue might also be present in android_core under the NodeMainExecutorService class. I will investigate that as well. Wanted to make sure we both aren't working on the same thing.

adamantivm, 2017-03-03:
@stratomda thanks very much for the work and for the heads-up. I haven't been able to carve out time to look into this so your pro-activity is appreciated. I'll check here before starting if I don't hear anything else by the time I do get time (not real soon).

closed 2017-03-06 by adamantivm