rosjava/rosjava_core #235: How to set build.gradle files (including build.gradle in app directory) in Android Studio to applying rosserial android?

frank0123, 2017-03-28:
Hi, guys! I'm new to ros and rosserial-android, recently I'm following the  Adam Stambler APIs and official guiding to built my rosserial-android demo in Android Studio, but there're a lots of mistakes in classes. I'm wonder it is the APIs are too old to follow or some configurations went wrong in my build.gradle files? I just know that, the rosserial-android need the ros-java core and ros-android core support, so how should I to set my build.gradle files to apply the rosserial-android?

adamantivm, 2017-03-28:
Hi Frank, could you post a link to the tutorials that you mention? What instructions are you trying to follow?

frank0123, 2017-03-28:
Here is the APIs link: http://docs.ros.org/fuerte/api/rosserial_android/html/annotated.html

And wiki of rosserial-android: http://wiki.ros.org/rosserial_android

jubeira, 2017-03-28:
Hi @frank0123 , 
The code for rosserial looks a bit old to me -it says it depends on Rosjava Fuerte, and the current version is Kinetic-. In this repo, as you can see, the oldest branch is Groovy, but I wouldn't rely on code that is that old.

I would try to make it work with the current release (tag 0.3.2, kinetic branch). You will probably need to revamp the code in rosserial to do so. To add the latest dependencies for ros-android and rosjava add the following lines to your build.gradle file:

```
  compile('org.ros.android_core:android_10:[0.3, 0.4)') {
    exclude group: 'junit'
    exclude group: 'xml-apis'
  }
```

frank0123, 2017-03-29:
@jubeira Thanks so much! I know that code is too old and useless to me, but the real problem is I can't get a correct configuration in build.gradle files with the support like core of ros_java/ros_android, and something else (rosserial_java/rosserial_msgs etc.) to making the rosserial_android works.  

So, although as you suggested put the latest dependencies of ros-android and ros-java in my build.gradle file, there's comes the error report: **Failed to resolve org.ros.android_core:android_10:[0.3, 0.4)**. I mean there're something about dependencies need to add in project build.gradle file, and some plugins(like apply plugin: 'ros_android') to initialise in build.gradle file at app directory. However, I never get a totally correct configuration plan until now. 

By the way, can you open the source site that official(ROS.org) provided:  https://kforge.ros.org/rosserial/android ? Sadly, I can't. If you can, could you show me the content in any way? Thanks in advance!

frank0123, 2017-03-29:
As following the guiding: http://wiki.ros.org/android/Tutorials/indigo/Installation%20-%20Android%20Studio%20Development%20Environment , here is my sub project's build.gradle file:

    buildscript {
        repositories {
            maven {
                url 'https://github.com/rosjava/rosjava_mvn_repo/raw/master'
            }
            mavenCentral()
            jcenter()
        }

        dependencies {
            classpath 'com.android.tools.build:gradle:2.2.3'
        }
    }

    allprojects {
        repositories {
            maven {
                url 'https://github.com/rosjava/rosjava_mvn_repo/raw/master'
            }
            mavenCentral()
            jcenter()
        }
    }

    task clean(type: Delete) {
        delete rootProject.buildDir
    }

and here is my app directory build.gradle file:

    apply plugin: 'com.android.application'

    android {
        compileSdkVersion 25
        buildToolsVersion "25.0.2"

        defaultConfig {
            applicationId "com.rosserial.rosserialAndroidDemo"
            minSdkVersion 15
            targetSdkVersion 25
            versionCode 1
            versionName "1.0"
            testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
        }

        buildTypes {
            release {
                minifyEnabled false
                proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
            }
        }
    }

    dependencies {
        compile fileTree(dir: 'libs', include: ['*.jar'])
        androidTestCompile('com.android.support.test.espresso:espresso-core:2.2.2', {
            exclude group: 'com.android.support', module: 'support-annotations'
        })

        compile 'com.android.support:appcompat-v7:25.3.0'
        compile 'com.android.support.constraint:constraint-layout:1.0.1'

        compile 'org.ros.android_core:android_gingerbread_mr1:[0.1,0.2)'
        compile 'org.ros.android_core:android_honeycomb_mr2:[0.1,0.2)'
        compile 'org.ros.rosjava_core:rosjava:[0.1,0.2)'
        compile 'org.ros.android_core:android_15:[0.3,0.4)'

        testCompile 'junit:junit:4.12'
    }

but there're somes an error: **Error:Execution failed for task ':app:prepareDebugAndroidTestDependencies'.**

jubeira, 2017-03-29:
Hi @frank0123 ,
First of all, no, I can't access the link you specified above (https://kforge.ros.org/rosserial/android) either.

What rosjava build tools version are you using? Are you in ROS Kinetic or Indigo? 
That tutorial seems a bit old. The easiest way of getting a project started in my opinion is using android-catkin packages, which create the project's build.gradle file pointing to [rosjava_mvn_repo](https://github.com/rosjava/rosjava_mvn_repo), and that will let you add the required dependencies.

If you tell me what ROS version and what rosjava version you are using, I can give you some pointers.

stratomda, 2017-03-29:
@frank0123 Do you need both android_gingerbread_mr1 and android_honeycomb_mr2? If not I would suggest removing. Also you do not need to specify rosjava compile, that will be pulled in by android_15. You need to add the following to your android_15 dependencies:
```gradle    
compile('org.ros.android_core:android_15:[0.3:0.4')
            {
                exclude group: 'junit'
                exclude group: 'xml-apis'
            }
```
Hope this helps.

frank0123, 2017-04-07:
Hey, guys! I'm so appreciated that you provided me so many suggestions, thanks so much. However, since the rossrial-android API is too old to follow, and there have been not come a good guiding for the rosserial-android beginner to follow and build an example-project like **Hello World**, so I'm going to give up the rosserial-android. The reason why I have to use rosserial-android is I need my application send the messages to ROS without the ROS-environment, so rosserial-android is the best way to get that. 

Fortunately, there's another way seems like the rosserial-android that communicate with the ROS without the ROS environment. Here is the ROS WiKi guiding link: http://wiki.ros.org/android/Tutorials/kinetic/Installation%20-%20Android%20Studio%20Development%20Environment . However there're some troubles when I'm follow the steps to set my build.gradle files, here is the error messages:

_**Error:Android tasks have already been created. This happens when calling android.applicationVariants, android.libraryVariants or android.testVariants. Once these methods are called, it is not possible to continue configuring the model.**_

And here is my build.gradle files:

`buildscript {`
      `apply from: "https://github.com/rosjava/android_core/raw/kinetic/buildscript.gradle"`
`}`

`allprojects {}`

`task clean(type: Delete) {`
      `delete rootProject.buildDir`
`}`

-----------------------------------------------------------------------------------
//app directory

`apply plugin: 'com.android.application'`
`apply plugin: 'ros-android'`

`android {`
    `compileSdkVersion 25`
    `buildToolsVersion "25.0.2"`

    `defaultConfig {`
        `applicationId "com.rosserial.rosserialEnviromentTest"`
        `minSdkVersion 15`
        `targetSdkVersion 25`
        `versionCode 1`
        `versionName "1.0"`
        `testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"`
    `}`

    `buildTypes {`
        `release {`
            `minifyEnabled false`
            `proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'`
        `}`
    `}`

    `// Exclude a few files that are duplicated across our dependencies and`
    `// prevent packaging Android applications.`
    `packagingOptions {
        `exclude "META-INF/LICENSE.txt"`
        `exclude "META-INF/NOTICE.txt"`
    `}`
`}`

`dependencies {`
    `compile fileTree(dir: 'libs', include: ['*.jar'])`

    `androidTestCompile('com.android.support.test.espresso:espresso-core:2.2.2', {`
        `exclude group: 'com.android.support', module: 'support-annotations'`
    `})`

    `compile 'com.android.support:appcompat-v7:25.3.1'`
    `compile 'com.android.support.constraint:constraint-layout:1.0.1'`

    `compile 'org.ros.android_core:android_10:[0.3,0.4)'`

    `testCompile 'junit:junit:4.12'`
`}`

I don' know where the problem is, do you guys have any ideas?

jubeira, 2017-04-07:
@frank0123 I think you are close to getting it right.
I suggest you to try the rosjava-catkin tools if you are using Kinetic. You can create an Android package and an Android project with those tools. The top level and module level scripts are generated automatically for you, and you will just need to add the dependencies in the module-level script.
Take a look at http://wiki.ros.org/android - building android-catkin packages. It will guide you to build a new Android package from the command line (the wiki is down ATM, I will send the tutorial's link when it's up).
Also, to add Android core you will need something like this (see [this issue](https://github.com/rosjava/android_core/issues/209)).
```
  compile('org.ros.android_core:android_10:[0.3, 0.4)') {
    exclude group: 'junit'
    exclude group: 'xml-apis'
  }
```

frank0123, 2017-04-08:
Well, now I'm following the guiding to build my ros environment at this link: [http://wiki.ros.org/android/Tutorials/kinetic/Installation%20-%20ROS%20Development%20Environment](url) . Unfortunately, I nearly can't make the first command line **_sudo apt-get install ros-kinetic-rosjava-build-tools_**. My system is **_Ubuntu 16.04(64-bit) LTS_**, and I had updated my **_apt-get_**, however there's come the error:

`E: Unable to locate package ros-kinetic-rosjava-build-tools`

What's wrong?

adamantivm, 2017-04-09:
Hi @frank0123 . You need to set-up the ROS repositories first, as explained here: http://wiki.ros.org/kinetic/Installation/Ubuntu

Also, I would suggest that for general support questions you use http://answers.ros.org instead of github issues, which are more oriented towards reporting concrete bugs or feature requests. You may get more answers here.

closed 2017-04-09 by adamantivm