rosjava/rosjava_core #237: DefaultPublisher Leaks DefaultNode

stratomda, 2017-03-29:
The DefaultPublisher class leaks the owning DefaultNode class. The corresponding TCP connection managers are also left open and unable to be garabage collected.  The fix is to shutdown the listeners in the DefaultPublisher and remove the publishers and subscribers from the DefaultNode's topicParticipantManager.

adamantivm, 2017-03-29:
Thanks for this analysis @stratomda . Do you think you'll be submitting a fix for this as well?

stratomda, 2017-03-29:
@adamantivm I have fixed this and tested in indigo. I will update in the kinetic branch for ease of merging the pull request.

adamantivm, 2017-03-29:
sounds great. Thank you very much!

closed 2017-03-30 by adamantivm

jubeira, 2017-03-30:
@adamantivm perhaps this could be cherrypicked to Indigo as well right?

adamantivm, 2017-03-30:
absolutely.