rosjava/rosjava_core #246: How does subscriber reconnect to publisher when disconnected?

jerryzwork, 2017-06-09:
I found that subscriber will reconnect publisher when disconnected in roscpp or rospy, but not in rosjava.

The "connect" method in TcpClient has no retry. How do I reconnect when disconnected? 

Thanks in advance!

jubeira, 2017-06-09:
Hi @jerryzwork! could you provide a little bit of context? When you say "when disconnected", you mean that the node connected to the topic is shut down? 

In general, the topic connections are created in the `onStart` method of a node, which is called by `NodeMainExecutor#execute`. If a node is shut down, it can be re-executed without creating the node object again; that will call `onStart` again and restore the connections.

jerryzwork, 2017-06-09:
I mean that the wifi had disconnected, the node had not shut down. How can I recover the connection between subscriber and publisher when wifi connected again?The netty likely had some solution for reconnect when listened the channel had closed.

jubeira, 2017-06-09:
I see.. I never tried that case in rosjava neither in rospy/cpp. Perhaps its not implemented in rosjava, and its a matter of using the netty functionality in `TcpClient`. 
If you try implementing it feel free to send a PR, I can help reviewing / testing if needed. Fixes / features like this one usually get merged pretty fast. It could be a good contribution for others too!

jerryzwork, 2017-06-09:
ok.I will try it recently.

closed 2017-06-09 by jerryzwork