rosjava/rosjava_core #247: Building error using the Gradle build system

ChrisYin, 2017-06-11:
I followed the install documentation: http://rosjava.github.io/rosjava_core/latest/building.html#gradleto install rosjava. When running this command

>  ./gradlew install 

Java send an error like these:

> Exception in thread "main" java.lang.RuntimeException: javax.net.ssl.SSLException: java.security.ProviderException: java.security.InvalidKeyException: EC parameters error
> 	at org.gradle.wrapper.ExclusiveFileAccessManager.access(ExclusiveFileAccessManager.java:78)
> 	at org.gradle.wrapper.Install.createDist(Install.java:44)
> 	at org.gradle.wrapper.WrapperExecutor.execute(WrapperExecutor.java:126)
> 	at org.gradle.wrapper.GradleWrapperMain.main(GradleWrapperMain.java:55)
> Caused by: javax.net.ssl.SSLException: java.security.ProviderException: java.security.InvalidKeyException: EC parameters error
> 	at sun.security.ssl.Alerts.getSSLException(Alerts.java:208)
> 	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1914)
> 	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1872)
> 	at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1855)
> 	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1376)
> 	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1353)
> 	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
> 	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
> 	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1348)
> 	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:254)
> 	at org.gradle.wrapper.Download.downloadInternal(Download.java:56)
> 	at org.gradle.wrapper.Download.download(Download.java:42)
> 	at org.gradle.wrapper.Install$1.call(Install.java:57)
> 	at org.gradle.wrapper.Install$1.call(Install.java:44)
> 	at org.gradle.wrapper.ExclusiveFileAccessManager.access(ExclusiveFileAccessManager.java:65)
> 	... 3 more
> Caused by: java.security.ProviderException: java.security.InvalidKeyException: EC parameters error
> 	at sun.security.pkcs11.P11Key$P11ECPublicKey.getEncodedInternal(P11Key.java:1024)
> 	at sun.security.pkcs11.P11Key.equals(P11Key.java:158)
> 	at java.util.ArrayList.indexOf(ArrayList.java:298)
> 	at java.util.ArrayList.contains(ArrayList.java:281)
> 	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:239)
> 	at sun.security.validator.Validator.validate(Validator.java:260)
> 	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:326)
> 	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:231)
> 	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:126)
> 	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1459)
> 	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:213)
> 	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:961)
> 	at sun.security.ssl.Handshaker.process_record(Handshaker.java:897)
> 	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1033)
> 	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1342)
> 	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1369)
> 	... 13 more
> Caused by: java.security.InvalidKeyException: EC parameters error
> 	at sun.security.ec.ECParameters.getAlgorithmParameters(ECParameters.java:284)
> 	at sun.security.ec.ECPublicKeyImpl.<init>(ECPublicKeyImpl.java:59)
> 	at sun.security.pkcs11.P11Key$P11ECPublicKey.getEncodedInternal(P11Key.java:1021)
> 	... 28 more
> Caused by: java.security.NoSuchProviderException: no such provider: SunEC
> 	at sun.security.jca.GetInstance.getService(GetInstance.java:83)
> 	at sun.security.jca.GetInstance.getInstance(GetInstance.java:206)
> 	at java.security.Security.getImpl(Security.java:697)
> 	at java.security.AlgorithmParameters.getInstance(AlgorithmParameters.java:199)
> 	at sun.security.ec.ECParameters.getAlgorithmParameters(ECParameters.java:279)
> 	... 30 more

My environment: Ubuntu 14.04 + ROS indigo

adamantivm, 2017-06-11:
Hi @ChrisYin . Could you please let me know how did you end up on this documentation? And also what exactly you're trying to achieve? That will let me better understand how to direct our work fixing the code and tutorials or what to recommend you in order to overcome this issue

ChrisYin, 2017-06-12:
Thanks for your reply. I have solved this problem. The reason is that I can not download "gradle-2.2-all.zip" when I using 

> ./gradlew install
So, I download it manually and copy it to the "gradle/wraper" directory. Then, it works.
My goal is construct communication between Android phone  and Raspberry Pi3 in ROS, so that I can subscribe IMU data of phone. Then I'm trying to build rosjava. However, I found it is troublesome to use rosjava or rosandroid. Maybe I should use TCP/IP socket.

adamantivm, 2017-06-12:
Thanks for the report @ChrisYin

You should be able to use the rosjava Android packages directly, it would be much easier than this.

Look into http://wiki.ros.org/android

closed 2017-06-12 by adamantivm