rosjava/rosjava_core #249: Connect a ROSJava node with ROS running on Docker

jabrena, 2017-06-20:
Hi,

I am trying to develop some ROS nodes for EV3 Brick and I have some mini problems with the communication between ROSJava and a ROSCore running on a Docker image.

In the past, I participated in the development of the support for NXT:
http://wiki.ros.org/nxt_lejos

In the development, I am using a ROS distro from a Docker Image:
http://wiki.ros.org/docker/Tutorials/Docker

``` 
➜  docker run -it ros 
root@8a35e37140d5:/# roscore
... logging to /root/.ros/log/b2d34ffc-560c-11e7-b8c7-0242ac110002/roslaunch-8a35e37140d5-35.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://8a35e37140d5:40671/
ros_comm version 1.12.7

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES

auto-starting new master
process[master]: started with pid [46]
ROS_MASTER_URI=http://8a35e37140d5:11311/

setting /run_id to b2d34ffc-560c-11e7-b8c7-0242ac110002
process[rosout-1]: started with pid [59]
started core service [/rosout]
```
After the execution of ROSCore, from localhost, I am executing a simple rosjava example which run 2 nodes: Talker & Listener:

``` java

import nodes.Listener;
import nodes.Talker;
import org.ros.RosCore;
import org.ros.node.DefaultNodeMainExecutor;
import org.ros.node.NodeConfiguration;
import org.ros.node.NodeMain;
import org.ros.node.NodeMainExecutor;

import java.net.InetAddress;
import java.net.UnknownHostException;
import java.util.concurrent.TimeUnit;

public class ROSJavaTest {

    private static RosCore mRosCore;

    public static void main(String [] args) throws UnknownHostException {

        mRosCore = RosCore.newPublic("6a36967c0c57", 11311);
        mRosCore.start();
        try {
            mRosCore.awaitStart(5, TimeUnit.SECONDS);
        } catch (InterruptedException e) {
            e.printStackTrace();
        }
        System.out.println("Ros core started");

        startChatter();

        //TODO Improve this way
        while (true) {}
    }

    private static void startChatter() throws UnknownHostException {

        NodeMainExecutor e = DefaultNodeMainExecutor.newDefault();

        System.out.println("Starting listener node...");
        NodeConfiguration listenerConfig = NodeConfiguration.newPrivate();
        //String host = InetAddress.getLocalHost().getHostAddress();
        //NodeConfiguration nodeConfiguration = NodeConfiguration.newPublic(host);
        listenerConfig.setMasterUri(mRosCore.getUri());
        listenerConfig.setNodeName("Listener");
        NodeMain listener = new Listener();
        e.execute(listener, listenerConfig);

        System.out.println("Starting talker node...");
	    NodeConfiguration talkerConfig = NodeConfiguration.newPrivate();
	    talkerConfig.setMasterUri(mRosCore.getUri());
	    talkerConfig.setNodeName("Talker");
	    NodeMain talker = new Talker();
	    e.execute(talker, talkerConfig);

    }

}
```

If the name of the host is: `http://8a35e37140d5:40671/` I am trying to connect to that image, but I receive some errors. in the connection:

``` 
Ros core started
Starting listener node...
Starting talker node...
jun 21, 2017 12:06:55 AM org.ros.internal.node.client.Registrar <init>
INFORMACIÓN: MasterXmlRpcEndpoint URI: http://8a35e37140d5:40671/
jun 21, 2017 12:06:55 AM org.ros.internal.node.client.Registrar <init>
INFORMACIÓN: MasterXmlRpcEndpoint URI: http://8a35e37140d5:40671/
jun 21, 2017 12:06:55 AM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Listener, http://127.0.0.1:57082/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jun 21, 2017 12:06:55 AM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://127.0.0.1:57083/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jun 21, 2017 12:06:55 AM org.ros.internal.node.client.Registrar callMaster
GRAVE: Exception caught while communicating with master.
java.lang.RuntimeException: java.net.UnknownHostException: 8a35e37140d5
	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:157)
	at com.sun.proxy.$Proxy0.registerPublisher(Unknown Source)
	at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
	at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
	at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
	at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
	at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
	at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
	at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.net.UnknownHostException: 8a35e37140d5
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:286)
	at org.apache.commons.httpclient.protocol.DefaultProtocolSocketFactory.createSocket(DefaultProtocolSocketFactory.java:80)
	at org.apache.commons.httpclient.protocol.ControllerThreadSocketFactory$1.doit(ControllerThreadSocketFactory.java:91)
	at org.apache.commons.httpclient.protocol.ControllerThreadSocketFactory$SocketTask.run(ControllerThreadSocketFactory.java:158)
	... 1 more

jun 21, 2017 12:06:55 AM org.ros.node.DefaultNodeMainExecutor$RegistrationListener onError
GRAVE: Node error.
java.lang.RuntimeException: java.net.UnknownHostException: 8a35e37140d5
	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:157)
	at com.sun.proxy.$Proxy1.hasParam(Unknown Source)
	at org.ros.internal.node.client.ParameterClient.hasParam(ParameterClient.java:116)
	at org.ros.internal.node.parameter.DefaultParameterTree.has(DefaultParameterTree.java:66)
	at org.ros.internal.node.DefaultNode.start(DefaultNode.java:197)
	at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85)
	at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.net.UnknownHostException: 8a35e37140d5
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:286)
	at org.apache.commons.httpclient.protocol.DefaultProtocolSocketFactory.createSocket(DefaultProtocolSocketFactory.java:80)
	at org.apache.commons.httpclient.protocol.ControllerThreadSocketFactory$1.doit(ControllerThreadSocketFactory.java:91)
	at org.apache.commons.httpclient.protocol.ControllerThreadSocketFactory$SocketTask.run(ControllerThreadSocketFactory.java:158)
	... 1 more
``` 
How to solve the problem?

How to get the ip of that image?

Many thanks in advance.

Cheers.

Juan Antonio

jubeira, 2017-06-21:
Hi @jabrena,
I think the problem is that the hostname cannot be resolved.
```
java.lang.RuntimeException: java.net.UnknownHostException: 8a35e37140d5
```
Even though the name seems correct, you have to make sure that the system running the java program knows how to map that hostname to an IP address. I'm not an expert with Docker, but I'd start looking there.

Apart from that, perhaps ROS Answers is a better place for questions like these.

adamantivm, 2017-06-21:
Additionally, I believe that by default Docker doesn't expose all the ports from inside the container to the host machine, unless you do mappings or network configuration specifically for this - doesn't look for your sample it was done.

In any case, this is for sure a docker / networking questions, unrelated with rosjava.

closed 2017-06-21 by adamantivm

jabrena, 2017-06-21:
Hi @adamantivm, @jubeira, 

you have reason, I think that the problem is about the Networking features of the Docker. I will ask help in `ROS Answers`.

Anyway, Many thanks

broesdecat, 2017-07-31:
Hi all,

Accidentally, I ran into almost the same problem here today, and I think @jubeira is correct in his assumption (and also related to #9). However, I am wondering whether this is not actually a rosjava issue instead of a ROS or networking one.

My setup:
A. Docker with ros:indigo, running on top of Ubuntu 16.04, static IP 172.31.1.151
B. Kuka sunrise cabinet, running Windows 7 embedded, static IP 172.31.1.147
RosJava is configured based on those IP addresses.

Using RosJava (through [iiwa_stack](https://github.com/SalvoVirga/iiwa_stack)), almost all communication went fine: bidirectional ros node and ros topic info, publishing from B to A.
Publishing from A to B failed, with as only info some debug output logging a SEVERE unknownhostexception for name "broes-desktop".
So apparently, somewhere in the process, the communication switches from IP addresses to hostnames.
I am not familiar enough with ROS protocols to know whether those protocols impose that switch, or whether this is a choice/bug of rosjava?
Anyway, having to configure a DNS server just to get this resolved for a statically configured network is not my preferred solution.

Best,
Broes

jubeira, 2017-09-27:
@broesdecat the best way to debug this is to get the information for each topic with `rostopic info`. That will tell you which nodes are publishing/ subscribing to the topic, and the way the nodes were registered to the master. For example:
```
rostopic info /rosout
Type: rosgraph_msgs/Log

Publishers: 
 * /rostopic_4328_1506514722375 (http://juan-HOTH:38993/)

Subscribers: 
 * /rosout (http://juan-HOTH:46598/)
```
In this case, the node `rostopic_4328_1506514722375` is publishing to `/rosout`, and that node was registered using my hostname, not my IP address. If I wanted to listen to that node from somewhere else (i.e. another node), I'd need to be able to resolve that hostname from the system that executes the listener node. If not, I will be able to subscribe to `rosout`, but I won't get any message from the anonymous node `rostopic_4328_1506514722375`. 

Does that make sense? In your case, how are your nodes being registered in the master?

jabrena, 2017-09-27:
For virtual environments, now I am using vagrant images. To open ports are easier than with docker.

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/393