rosjava/rosjava_core #250: Cannot resolve dependencies on Apache commons libraries

ex-ratt, 2017-06-21:
After few months, I wanted to build one of my projects again after some changes. It uses rosjava, but can't resolve the dependencies on Apache commons libraries. A short excerpt:

```
Could not resolve all dependencies for configuration ':compileClasspath'.
> Could not find org.apache.commons:com.springsource.org.apache.commons.logging:1.1.1.
  Searched in the following locations:
      file:/home/ex-ratt/.m2/repository/org/apache/commons/com.springsource.org.apache.commons.logging/1.1.1/com.springsource.org.apache.commons.logging-1.1.1.pom
      file:/home/ex-ratt/.m2/repository/org/apache/commons/com.springsource.org.apache.commons.logging/1.1.1/com.springsource.org.apache.commons.logging-1.1.1.jar
      https://repo1.maven.org/maven2/org/apache/commons/com.springsource.org.apache.commons.logging/1.1.1/com.springsource.org.apache.commons.logging-1.1.1.pom
      https://repo1.maven.org/maven2/org/apache/commons/com.springsource.org.apache.commons.logging/1.1.1/com.springsource.org.apache.commons.logging-1.1.1.jar
      https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/apache/commons/com.springsource.org.apache.commons.logging/1.1.1/com.springsource.org.apache.commons.logging-1.1.1.pom
      https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/apache/commons/com.springsource.org.apache.commons.logging/1.1.1/com.springsource.org.apache.commons.logging-1.1.1.jar
  Required by:
      project : > org.ros.rosjava_core:rosjava:0.2.2
      project : > org.ros.rosjava_core:rosjava:0.2.2 > org.ros.rosjava_core:apache_xmlrpc_server:0.2.2
```

I had a look into the central Maven repository and indeed, there is no directory `https://repo1.maven.org/maven2/org/apache/commons/com.springsource.org.apache.commons.logging/`. Instead, the libraries `commons-logging`, `commons-net`, `commons-httpclient`, `commons-codec`, `commons-io`, and `commons-lang` can be found at the root-level of the repository, e.g. `https://repo1.maven.org/maven2/commons-logging/`.

I have no idea why that is and when it was changed, but it seems like you have to update all Gradle build files that declare dependencies on one of those Apache commons libraries. Otherwise, they cannot be found on systems where they do not reside in you local Gradle cache already.

jabrena, 2017-06-21:
Hi @ex-ratt,

Review this file:
https://github.com/ev3dev-lang-java/examples/blob/rosjava-helloworld/rosjava-helloworld/build.gradle

I fix the dependency problems.

Juan Antonio

ex-ratt, 2017-06-21:
Your project does not run into this issue, because you make use of the jcenter repository, where those Apache commons libraries are still available under the path that is required by rosjava. I just cloned your project to a new user account on my PC and called `./gradlew dependencies` to see where the jars would be downloaded from. And now that they are in my local gradle cache, they won't need downloading again.

The rosjava installation tutorial however does not mention jcenter and relies solely on Maven central. New users of rosjava therefore probably will run into this issue.

adamantivm, 2017-06-21:
Hi @ex-ratt . Thank you for reporting this. In order to help us update the documentation and make it easier for others, would you mind letting us know which rosjava installation tutorial were you using which doesn't specify jcenter?

ex-ratt, 2017-06-21:
I used the tutorials from the ROS wiki, specifically the no ROS installation, as I just need the jars in my Java projects:
http://wiki.ros.org/rosjava/Tutorials/indigo/No%20Ros%20Installation
http://wiki.ros.org/rosjava/Tutorials/kinetic/No%20Ros%20Installation

Just for clarification: Do you think that I should solve this issue by specifying the jcenter repository? If so, do you happen to know why some of the Apache commons libraries are not available anymore in the central Maven repository under the name specified in the rosjava gradle build files?

adamantivm, 2017-06-21:
Thank you @ex-ratt . I will review and update the tutorials if necessary.

Indeed the proper solution is to use jcenter.

In the past we had some artifacts which have their permanent home in jcenter manually mirrored to the rosjava maven repository, but this ended up being problematic because the rosjava maintainers over time didn't have the bandwidth to keep updating the maven repository to keep up with jcenter. So we now opted to have developers use the original repository source of dependencies, like it is done normally in other Java projects. See https://github.com/rosjava/rosjava_mvn_repo/pull/20

hawkina, 2017-08-09:
In my case it worked to set the $ROS_MAVEN_REPOSITORY to the indigo branch https://github.com/rosjava/rosjava_mvn_repo/raw/indigo . 

I did try to use jcenter, but that didn't work for me. Either I missed something or the paths are different there so that I would have to change all of the dependencies of the package. So... let's just say I'm happy it is building again.

jubeira, 2017-08-09:
@ex-ratt , @hawkina what is the rosjava artifact version you are targeting? Have you tried upgrading to `[0.3, 0.4)`? Those are the artifacts corresponding to the `kinetic` branch, where the dependencies were removed. The APIs are basically the same; most likely changing this won't break your code.
I've been using rosjava `0.3` and ahead for a while with Gradle and also with plain Maven, and I didn't experience these issues.

ex-ratt, 2017-08-10:
I was still targeting ROS Indigo and therefore used version 0.2+. Haven't tried the newer versions yet.

drigz, 2018-02-26:
@ex-ratt The [tutorial](http://wiki.ros.org/rosjava/Tutorials/kinetic/No%20Ros%20Installation#Gradle) now includes the jcenter repository. Does this resolve the issue?

```
repositories {
  maven {
    url "https://github.com/rosjava/rosjava_mvn_repo/raw/master"
  }
  mavenLocal()
  jcenter()
}
```

ex-ratt, 2018-02-27:
Using the jcenter repository resolved this issue for me last year. Thanks for updating the tutorial :)

closed 2018-02-27 by ex-ratt