rosjava/rosjava_core #251: ROSJava node listener registered doesn´t receive events.

jabrena, 2017-06-25:
Hi,

I continued with the issue reported: https://github.com/rosjava/rosjava_core/issues/249 with a physical ROS Kinect instalation. 

In the example:
https://github.com/ev3dev-lang-java/examples/blob/rosjava-helloworld/rosjava-helloworld/src/main/java/ListenerTest.java
https://github.com/ev3dev-lang-java/examples/blob/rosjava-helloworld/rosjava-helloworld/src/main/java/nodes/Listener.java

I register a Listener node from my laptop to the host running a ROSMaster:

```
Ros core started
Starting listener node...
jun 25, 2017 4:29:31 PM org.ros.internal.node.client.Registrar <init>
INFORMACIÓN: MasterXmlRpcEndpoint URI: http://192.168.1.244:11311/
jun 25, 2017 4:29:31 PM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Listener, http://127.0.0.1:64158/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jun 25, 2017 4:29:31 PM org.ros.internal.node.client.Registrar callMaster
INFORMACIÓN: Response<Success, Registered [/Listener] as publisher of [/rosout], [http://ev3dev:33975/]>
jun 25, 2017 4:29:31 PM org.ros.internal.node.topic.DefaultPublisher$1 onMasterRegistrationSuccess
INFORMACIÓN: Publisher registered: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Listener, http://127.0.0.1:64158/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jun 25, 2017 4:29:31 PM org.ros.internal.node.client.Registrar onSubscriberAdded
INFORMACIÓN: Registering subscriber: Subscriber<Topic<TopicIdentifier</chatter>, TopicDescription<std_msgs/String, 992ce8a1687cec8c8bd883ec73ca41d1>>>
jun 25, 2017 4:29:31 PM org.ros.internal.node.client.Registrar callMaster
INFORMACIÓN: Response<Success, Subscribed to [/chatter], []>
jun 25, 2017 4:29:31 PM org.ros.internal.node.topic.DefaultSubscriber$1 onMasterRegistrationSuccess
INFORMACIÓN: Subscriber registered: Subscriber<Topic<TopicIdentifier</chatter>, TopicDescription<std_msgs/String, 992ce8a1687cec8c8bd883ec73ca41d1>>>
``` 

The node is registered:

``` 
robot@ev3dev:~$ rostopic info /chatter
Type: std_msgs/String

Publishers: None

Subscribers: 
 * /Listener (http://127.0.0.1:64158/)
``` 

Later, from the terminal I publish some messages manually with the command: `rostopic`:

``` 
robot@ev3dev:~$ rostopic pub -r 10 /chatter  std_msgs/String "hello"
```

But in the Laptop side, I don´t receive any message from the event:

```
    @Override
    public void onStart(ConnectedNode connectedNode) {
        Subscriber<std_msgs.String> subscriber = connectedNode.newSubscriber("chatter", std_msgs.String._TYPE);
        subscriber.addMessageListener(new MessageListener<std_msgs.String>() {
            @Override
            public void onNewMessage(std_msgs.String message) {
                System.out.println("I heard: \"" + message.getData() + "\"");
            }
        });
    }
```

Any idea?

I think that the code is simple.

Why the listener node apper with the ip: `127.0.0.1`and not with the real network ip? Is it necessary to maintain in that side or it is possible to configure in some way?

Many thanks in advance.

Juan Antonio

adamantivm, 2017-06-26:
Hi @jabrena , the problem is most likely on [this line](https://github.com/ev3dev-lang-java/examples/blob/rosjava-helloworld/rosjava-helloworld/src/main/java/ListenerTest.java#L50). When you say "newPrivate" you're saying "create a new 'private network' node", thus choosing an IP address of 127.0.0.1 for the node.
Try using "newPublic" instead.

jabrena, 2017-06-26:
Hi @adamantivm, you had reason but now, appear another problem: From a Laptop and from a Raspberry Pi, the node run in the expected way but from a EV3 Brick the ROS Node remain in a determined state. I will explain in detail:

**From a laptop:**

When I execute the node Talker:
https://github.com/ev3dev-lang-java/examples/blob/rosjava-helloworld/rosjava-helloworld/src/main/java/TalkerTest.java

And the node is registered in `roscore`, the node start emitting messages to the topic `/chatter`

``` 
Ros core started
Starting talker node...
jun 26, 2017 8:50:16 PM org.ros.internal.node.client.Registrar <init>
INFORMACIÓN: MasterXmlRpcEndpoint URI: http://192.168.1.244:11311/
jun 26, 2017 8:50:16 PM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.198:55780/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jun 26, 2017 8:50:18 PM org.ros.internal.node.client.Registrar callMaster
INFORMACIÓN: Response<Success, Registered [/Talker] as publisher of [/rosout], [http://ev3dev:44575/]>
jun 26, 2017 8:50:18 PM org.ros.internal.node.topic.DefaultPublisher$1 onMasterRegistrationSuccess
INFORMACIÓN: Publisher registered: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.198:55780/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jun 26, 2017 8:50:18 PM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.198:55780/>, TopicIdentifier</chatter>>, Topic<TopicIdentifier</chatter>, TopicDescription<std_msgs/String, 992ce8a1687cec8c8bd883ec73ca41d1>>>>
0
jun 26, 2017 8:50:18 PM org.ros.internal.node.client.Registrar callMaster
INFORMACIÓN: Response<Success, Registered [/Talker] as publisher of [/chatter], [http://ev3dev:42089/]>
jun 26, 2017 8:50:18 PM org.ros.internal.node.topic.DefaultPublisher$1 onMasterRegistrationSuccess
INFORMACIÓN: Publisher registered: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.198:55780/>, TopicIdentifier</chatter>>, Topic<TopicIdentifier</chatter>, TopicDescription<std_msgs/String, 992ce8a1687cec8c8bd883ec73ca41d1>>>>
1
```

In the roscore side, it is possible to verify with some commands:

``` 
rostopic echo /chatter
``` 

**From a EV3 Brick:**

But, when I execute from the Brick, the node register but from my perspective something stop internally and the ROSNode is not able to change the state.

The traces are:

``` 
java -server -jar /home/robot/rosjava-helloworld-0.1.0-SNAPSHOT-all.jar 192.168.1.244
ev3dev#1|192.168.1.244
ev3dev#1|Ros core started
ev3dev#1|Starting talker node...
ev3dev#1|Jun 26, 2017 7:55:53 PM org.ros.internal.node.client.Registrar <init>
ev3dev#1|INFO: MasterXmlRpcEndpoint URI: http://192.168.1.244:11311/
ev3dev#1|Jun 26, 2017 7:56:10 PM org.ros.internal.node.client.Registrar onPublisherAdded
ev3dev#1|INFO: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.206:44235/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
ev3dev#1|Jun 26, 2017 7:56:21 PM org.ros.internal.node.client.Registrar callMaster
ev3dev#1|SEVERE: Exception caught while communicating with master.
ev3dev#1|org.ros.internal.node.xmlrpc.XmlRpcTimeoutException: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
ev3dev#1|	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:140)
ev3dev#1|	at com.sun.proxy.$Proxy0.registerPublisher(Unknown Source)
ev3dev#1|	at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
ev3dev#1|	at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
ev3dev#1|	at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
ev3dev#1|	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
ev3dev#1|	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
ev3dev#1|	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
ev3dev#1|	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
ev3dev#1|	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
ev3dev#1|	at java.lang.Thread.run(Thread.java:744)
ev3dev#1|Caused by: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
[... 42 lines cut ...]
ev3dev#1|INFO: MasterXmlRpcEndpoint URI: http://192.168.1.244:11311/
ev3dev#1|Jun 26, 2017 8:08:50 PM org.ros.internal.node.client.Registrar onPublisherAdded
[... cut ...]
ev3dev#1|Jun 26, 2017 8:08:51 PM org.ros.internal.node.topic.DefaultPublisher$1 onMasterRegistrationSuccess
ev3dev#1|INFO: Publisher registered: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.143:35687/>, TopicIdentifier</chatter>>, Topic<TopicIdentifier</chatter>, TopicDescription<std_msgs/String, 992ce8a1687cec8c8bd883ec73ca41d1>>>>
ev3dev#1|1
ev3dev#1|2
ev3dev#1|3
ev3dev#1|4
```

Another question, how to set the log messages to `TRACES` level to see more details?

Many thanks in advance.

Juan Antonio

adamantivm, 2017-06-27:
@jabrena I am sorry I don't have very concrete answers for you at this point.
I can't tell from the logs where the problem may be. If you say it works from the raspberry pi node, then it probably is not a rosjava problem but most likely again a networking problem.
You can diagnose the issue a little more by using "rostopic info" and "rosnode info" on the Linux box.

Regarding the logging question, I wish I was more familiar with it to be able to tell you exactly but I'm not. What I can tell you is that the log system in rosjava is using Apache Commons Logging, so you could try to increase the output log level by reading about that and trying to figure out how to configure it for more verbose logging output.

closed 2017-06-27 by adamantivm

jubeira, 2017-06-27:
@jabrena what are the IP addresses of the two devices you are trying to connect? Remember that registering a topic to the Master node is not enough: if a node publishes to a topic and another one subscribes to the same topic, those two nodes need to be able to communicate (i.e. resolve IP addresses and/or hostnames, depending how the nodes were started).

Take a look here: http://wiki.ros.org/ROS/Technical%20Overview#Establishing_a_topic_connection.
This applies to ROS and ROSJava as well.

jabrena, 2017-06-27:
Hi @jubeira, your comment has sense.

Today, I tested from another brick with the same result. I will explain in detail the architecture of the POC:

- 192.168.1.244: ROSCore
- 192.168.1.143: Listener running on Raspberry Pi
- 192.168.1.198: Taker running on a laptop
- 192.168.1.206: Brick with Talker
- 192.168.1.85: Brick 2 with Talker

Note: In every test, I only run one talker and one listener.

In the last try, I am listening from the laptop. The Brick 2 with the example: `Talker` start running with the same behaviour than other days:

``` 
java -server -jar /home/robot/rosjava-helloworld-0.1.0-SNAPSHOT-all.jar 192.168.1.244
ev3dev#1|192.168.1.244
ev3dev#1|Ros core started
ev3dev#1|Starting talker node...
ev3dev#1|Jun 27, 2017 7:19:17 PM org.ros.internal.node.client.Registrar <init>
ev3dev#1|INFO: MasterXmlRpcEndpoint URI: http://192.168.1.244:11311/
ev3dev#1|Jun 27, 2017 7:19:32 PM org.ros.internal.node.client.Registrar onPublisherAdded
ev3dev#1|INFO: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.85:36512/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
ev3dev#1|Jun 27, 2017 7:19:43 PM org.ros.internal.node.client.Registrar callMaster
ev3dev#1|SEVERE: Exception caught while communicating with master.
ev3dev#1|org.ros.internal.node.xmlrpc.XmlRpcTimeoutException: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
ev3dev#1|	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:140)
ev3dev#1|	at com.sun.proxy.$Proxy0.registerPublisher(Unknown Source)
ev3dev#1|	at org.ros.internal.node.client.MasterClient.registerPublisher(MasterClient.java:145)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:138)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1$1.call(Registrar.java:135)
ev3dev#1|	at org.ros.internal.node.client.Registrar.callMaster(Registrar.java:111)
ev3dev#1|	at org.ros.internal.node.client.Registrar.access$100(Registrar.java:51)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1.call(Registrar.java:135)
ev3dev#1|	at org.ros.internal.node.client.Registrar$1.call(Registrar.java:132)
ev3dev#1|	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
ev3dev#1|	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
ev3dev#1|	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
ev3dev#1|	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
ev3dev#1|	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
ev3dev#1|	at java.lang.Thread.run(Thread.java:744)
ev3dev#1|Caused by: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
ev3dev#1|	at org.apache.xmlrpc.client.TimingOutCallback.waitForResponse(TimingOutCallback.java:77)
ev3dev#1|	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:138)
ev3dev#1|	... 14 more
ev3dev#1|
ev3dev#1|Jun 27, 2017 7:19:43 PM org.ros.node.DefaultNodeMainExecutor$RegistrationListener onError
ev3dev#1|SEVERE: Node error.
ev3dev#1|org.ros.internal.node.xmlrpc.XmlRpcTimeoutException: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
ev3dev#1|	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:140)
ev3dev#1|	at com.sun.proxy.$Proxy1.hasParam(Unknown Source)
ev3dev#1|	at org.ros.internal.node.client.ParameterClient.hasParam(ParameterClient.java:116)
ev3dev#1|	at org.ros.internal.node.parameter.DefaultParameterTree.has(DefaultParameterTree.java:66)
ev3dev#1|	at org.ros.internal.node.DefaultNode.start(DefaultNode.java:197)
ev3dev#1|	at org.ros.internal.node.DefaultNode.access$000(DefaultNode.java:85)
ev3dev#1|	at org.ros.internal.node.DefaultNode$1.run(DefaultNode.java:173)
ev3dev#1|	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
ev3dev#1|	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
ev3dev#1|	at java.lang.Thread.run(Thread.java:744)
ev3dev#1|Caused by: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
ev3dev#1|	at org.apache.xmlrpc.client.TimingOutCallback.waitForResponse(TimingOutCallback.java:77)
ev3dev#1|	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:138)
ev3dev#1|	... 9 more
ev3dev#1|
ev3dev#1|Jun 27, 2017 7:19:44 PM org.ros.internal.node.topic.DefaultPublisher$1 onMasterRegistrationFailure
ev3dev#1|INFO: Publisher registration failed: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.85:36512/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
ev3dev#1|Jun 27, 2017 7:20:00 PM org.ros.internal.node.client.Registrar callMaster
ev3dev#1|SEVERE: Exception caught while communicating with master.
ev3dev#1|org.ros.internal.node.xmlrpc.XmlRpcTimeoutException: org.apache.xmlrpc.client.TimingOutCallback$TimeoutException: No response after waiting for 10000 milliseconds.
ev3dev#1|	at org.ros.internal.node.xmlrpc.XmlRpcClientFactory$1.invoke(XmlRpcClientFactory.java:140)
[... 99 lines cut ...]
```

[... cut ...]
ev3dev#1|SEVERE: org.apache.xmlrpc.webserver.Connection$BadRequestException: GET
ev3dev#1|Jun 27, 2017 7:32:48 PM org.apache.xmlrpc.server.XmlRpcErrorLogger log
ev3dev#1|SEVERE: org.apache.xmlrpc.webserver.Connection$BadRequestException: GET
```

So, I think that the Node on Brick2, at the end didn´t register on ROS and it continue in some state that I don´t understand.

I think that this issue should be opened in my humble opinion.

Questions:

+ How to solve the problem? How to define more time to register the node?
+ How to retry the process?

[image]

jubeira, 2017-06-27:
@jabrena I see you are struggling, but believe me, this is most likely a networking problem and not a rosjava one.
If you see the first block of logs at the bottom: 
```
ev3dev#1|INFO: Publisher registration failed: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.85:36512/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
ev3dev#1|Jun 27, 2017 7:20:08 PM org.ros.internal.node.client.Registrar callMaster
ev3dev#1|INFO: Response<Success, Registered [/Talker] as publisher of [/rosout], [http://ev3dev:40358/]>
ev3dev#1|Jun 27, 2017 7:20:09 PM org.ros.internal.node.topic.DefaultPublisher$1 onMasterRegistrationSuccess
ev3dev#1|INFO: Publisher registered: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://192.168.1.85:36512/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
```

It says that the talker at .1.85 failed to connect (you can see the exceptions above, that shouldn't be happening). It also says that the node `Talker` was registered as a publisher of `rosout`. Have you tried logging things from the `Talker` node and echoing `rosout`? You can also try `rostopic info rosout` to see if it has publishers.

The problem is between the system running the roscore at .244 and the brick at .85. The timeout is 10 seconds, and the ROS Master is not answering a publisher "advertise" request. I don't think that adding more time will solve your problem. How about using Wireshark to see if the request is actually arriving to the ROS master? Can you ping the brick from the ROS master and vice versa? (the log above pings another IP address, not the one at .85).

Perhaps try running the ROS master somewhere else, at least to try if it works. The talker/listener example is pretty simple and should work fine.

adamantivm, 2017-06-27:
>  ev3dev#1|INFO: Response<Success, Registered [/Talker] as publisher of [/rosout], [http://ev3dev:40358/]>

That line specifically is saying that it will try to connect to the hostname `ev3dev`, with that name (not IP address). Can you ping ev3dev successfully by name from all the other nodes?

Perhaps one thing you can try, to make absolutely sure, is to list all node names in each other's `/etc/hosts` files and make sure you can ping each other node via hostname too.

jabrena, 2017-06-27:
Hi @adamantivm, you had reason:

```
robot@brick1:~$ ping ev3dev
PING ev3dev (192.168.1.143) 56(84) bytes of data.
From brick1 (192.168.1.206) icmp_seq=1 Destination Host Unreachable
From brick1 (192.168.1.206) icmp_seq=2 Destination Host Unreachable
From brick1 (192.168.1.206) icmp_seq=3 Destination Host Unreachable
From brick1 (192.168.1.206) icmp_seq=4 Destination Host Unreachable
``` 

But I edit `/etc/hosts` and now from the brick1 (192.168.1.206) I connect:

``` 
robot@brick1:/etc$ ping ev3dev
PING ev3dev (192.168.1.244) 56(84) bytes of data.
64 bytes from ev3dev (192.168.1.244): icmp_seq=1 ttl=64 time=13.2 ms
64 bytes from ev3dev (192.168.1.244): icmp_seq=2 ttl=64 time=11.5 ms
64 bytes from ev3dev (192.168.1.244): icmp_seq=3 ttl=64 time=11.7 ms
64 bytes from ev3dev (192.168.1.244): icmp_seq=4 ttl=64 time=12.0 ms
64 bytes from ev3dev (192.168.1.244): icmp_seq=5 ttl=64 time=13.1 ms
```

@jubeira, Yes, from roscore, the node talker is registered:

```
robot@ev3dev:~$ rostopic info rosout
Type: rosgraph_msgs/Log

Publishers: 
 * /Listener (http://192.168.1.198:60770/)
 * /Talker (http://192.168.1.206:32935/)

Subscribers: 
 * /rosout (http://ev3dev:40358/)

robot@ev3dev:~$ 
```

Besides, I retrieve the info about the node Talker:

``` 
robot@ev3dev:~$ rosnode list
/Listener
/Talker
/rosout
robot@ev3dev:~$ rosnode
rosnode is a command-line tool for printing information about ROS Nodes.

Commands:
	rosnode ping	test connectivity to node
	rosnode list	list active nodes
	rosnode info	print information about node
	rosnode machine	list nodes running on a particular machine or list machines
	rosnode kill	kill a running node
	rosnode cleanup	purge registration information of unreachable nodes

Type rosnode <command> -h for more detailed usage, e.g. 'rosnode ping -h'

robot@ev3dev:~$ rosnode ping /Talker
rosnode: node is [/Talker]
pinging /Talker with a timeout of 3.0s
xmlrpc reply from http://192.168.1.206:32935/	time=561.779976ms
xmlrpc reply from http://192.168.1.206:32935/	time=221.433878ms
xmlrpc reply from http://192.168.1.206:32935/	time=212.157011ms
xmlrpc reply from http://192.168.1.206:32935/	time=199.072123ms
^Cping average: 298.610747ms
robot@ev3dev:~$ rosnode info /Talker
--------------------------------------------------------------------------------
Node [/Talker]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: None

Services: None

contacting node http://192.168.1.206:32935/ ...
Pid: 609
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS

robot@ev3dev:~$ 
```

jabrena, 2017-06-27:
The doubt is why Talker:
https://github.com/ev3dev-lang-java/examples/blob/rosjava-helloworld/rosjava-helloworld/src/main/java/nodes/Talker.java

```  java
public class Talker extends AbstractNodeMain {

    @Override
    public GraphName getDefaultNodeName() {
        return GraphName.of("Talker");
    }

    @Override
    public void onStart(ConnectedNode connectedNode) {

        final Publisher<std_msgs.String> publisher =
                connectedNode.newPublisher("chatter", std_msgs.String._TYPE);

        connectedNode.executeCancellableLoop(new CancellableLoop() {
            private int sequence = 0;

            @Override
            protected void setup() {
                sequence = 0;
            }

            @Override
            protected void loop() throws InterruptedException {
                System.out.println(sequence);
                std_msgs.String str = publisher.newMessage();
                str.setData("Hello World" + sequence);
                publisher.publish(str);
                sequence++;
                Thread.sleep(1000);
            }

        });
    }

}
``` 

is pointing to `/rosout` and not `/chatter`?

``` 
Node [/Talker]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: None

Services: None

contacting node http://192.168.1.206:32935/ ...
Pid: 609
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
```

jabrena, 2017-06-27:
/etc/hosts about all machines:

**Brick1 (192.168.1.206)**

```
robot@brick1:~$ cat /etc/hosts
127.0.0.1	localhost
::1	localhost ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters
172.17.0.2	0b32e9ef465e
192.168.1.244	pistorms1
192.168.1.198	soulFactory
``` 

**Pistorms1 (192.168.1.244)**

``` 
robot@pistorms1:~$ cat /etc/hosts
127.0.0.1	localhost
::1	localhost ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters
172.17.0.3	7e142f3c7b64
192.168.1.206	brick1
192.168.1.198	soulFactory
``` 

**OSX (192.168.1.198)**

```
➜  cat /etc/hosts
##
# Host Database
#
# localhost is used to configure the loopback interface
# when the system is booting.  Do not change this entry.
##
127.0.0.1	localhost
255.255.255.255	broadcasthost
::1             localhost
192.168.1.244 	pistorms1
192.168.1.206	brick1
```

jabrena, 2017-06-27:
```
robot@pistorms1:~$ export ROS_IP=192.168.1.244
robot@pistorms1:~$ roscore
... logging to /home/robot/.ros/log/a7798982-5b89-11e7-aaeb-b827eb285ab4/roslaunch-pistorms1-1191.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.244:34267/
ros_comm version 1.12.7

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES

auto-starting new master
process[master]: started with pid [1202]
ROS_MASTER_URI=http://192.168.1.244:11311/

setting /run_id to a7798982-5b89-11e7-aaeb-b827eb285ab4
process[rosout-1]: started with pid [1215]
started core service [/rosout]
```

But the problem continues.

jabrena, 2017-06-28:
I am testing with `Wireshark` as @jubeira suggested.

jabrena, 2017-07-02:
Hi @adamantivm, @jubeira finally, I could configure logs to show all information about the execution.

This is the output from a right execution from the PC:

```
jul 02, 2017 8:12:48 PM org.jboss.netty.channel.socket.nio.NioProviderMetadata
DETALLADO: Couldn't determine the NIO constraint level from the system properties; using the safest level (2)
jul 02, 2017 8:12:49 PM org.ros.internal.node.client.Registrar <init>
INFORMACIÓN: MasterXmlRpcEndpoint URI: http://10.0.1.2:11311/
jul 02, 2017 8:12:49 PM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Listener, http://soulFactory:62503/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Java version: 1.8.0_121
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Java vendor: Oracle Corporation
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Java class path: rosjava-helloworld-0.1.0-SNAPSHOT-all.jar
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Operating system name: Mac OS X
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Operating system architecture: x86_64
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Operating system version: 10.12.5
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SUN 1.8: SUN (DSA key/parameter generation; DSA signing; SHA-1, MD5 digests; SecureRandom; X.509 certificates; JKS & DKS keystores; PKIX CertPathValidator; PKIX CertPathBuilder; LDAP, Collection CertStores, JavaPolicy Policy; JavaLoginConfig Configuration)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunRsaSign 1.8: Sun RSA signature provider
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunEC 1.8: Sun Elliptic Curve provider (EC, ECDSA, ECDH)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunJSSE 1.8: Sun JSSE provider(PKCS12, SunX509/PKIX key/trust factories, SSLv3/TLSv1/TLSv1.1/TLSv1.2)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunJCE 1.8: SunJCE Provider (implements RSA, DES, Triple DES, AES, Blowfish, ARCFOUR, RC2, PBE, Diffie-Hellman, HMAC)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunJGSS 1.8: Sun (Kerberos v5, SPNEGO)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunSASL 1.8: Sun SASL provider(implements client mechanisms for: DIGEST-MD5, GSSAPI, EXTERNAL, PLAIN, CRAM-MD5, NTLM; server mechanisms for: DIGEST-MD5, GSSAPI, CRAM-MD5, NTLM)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: XMLDSig 1.8: XMLDSig (DOM XMLSignatureFactory; DOM KeyInfoFactory; C14N 1.0, C14N 1.1, Exclusive C14N, Base64, Enveloped, XPath, XPath2, XSLT TransformServices)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunPCSC 1.8: Sun PC/SC provider
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Apple 1.8: Apple Provider
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.useragent = Jakarta Commons-HttpClient/3.1
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.version = HTTP/1.1
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.connection-manager.class = class org.apache.commons.httpclient.SimpleHttpConnectionManager
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.cookie-policy = default
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.element-charset = US-ASCII
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.content-charset = ISO-8859-1
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.method.retry-handler = org.apache.commons.httpclient.DefaultHttpMethodRetryHandler@c18a75e
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.dateparser.patterns = [EEE, dd MMM yyyy HH:mm:ss zzz, EEEE, dd-MMM-yy HH:mm:ss zzz, EEE MMM d HH:mm:ss yyyy, EEE, dd-MMM-yyyy HH:mm:ss z, EEE, dd-MMM-yyyy HH-mm-ss z, EEE, dd MMM yy HH:mm:ss z, EEE dd-MMM-yyyy HH:mm:ss z, EEE dd MMM yyyy HH:mm:ss z, EEE dd-MMM-yyyy HH-mm-ss z, EEE dd-MMM-yy HH:mm:ss z, EEE dd MMM yy HH:mm:ss z, EEE,dd-MMM-yy HH:mm:ss z, EEE,dd-MMM-yyyy HH:mm:ss z, EEE, dd-MM-yyyy HH:mm:ss z]
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.connection.timeout = 60000
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.connection.timeout = 60000
[... cut ...]
	at org.apache.commons.httpclient.protocol.ControllerThreadSocketFactory$SocketTask.run(ControllerThreadSocketFactory.java:158)
	... 1 more

jul 02, 2017 7:32:01 PM org.ros.internal.node.topic.DefaultPublisher onMasterRegistrationFailure
INFORMACIÓN: Publisher registration failed: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Talker, http://ev3dev:62340/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
```

jabrena, 2017-07-02:
If you observe, the Brick connect with roscore, but it is pretty interesting if you compare the execution in both sides:

PC with fast CPU:

```
jul 02, 2017 8:12:48 PM org.jboss.netty.channel.socket.nio.NioProviderMetadata
DETALLADO: Couldn't determine the NIO constraint level from the system properties; using the safest level (2)
jul 02, 2017 8:12:49 PM org.ros.internal.node.client.Registrar <init>
INFORMACIÓN: MasterXmlRpcEndpoint URI: http://10.0.1.2:11311/
jul 02, 2017 8:12:49 PM org.ros.internal.node.client.Registrar onPublisherAdded
INFORMACIÓN: Registering publisher: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Listener, http://soulFactory:62503/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Java version: 1.8.0_121
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Java vendor: Oracle Corporation
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Java class path: rosjava-helloworld-0.1.0-SNAPSHOT-all.jar
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Operating system name: Mac OS X
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Operating system architecture: x86_64
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Operating system version: 10.12.5
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SUN 1.8: SUN (DSA key/parameter generation; DSA signing; SHA-1, MD5 digests; SecureRandom; X.509 certificates; JKS & DKS keystores; PKIX CertPathValidator; PKIX CertPathBuilder; LDAP, Collection CertStores, JavaPolicy Policy; JavaLoginConfig Configuration)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunRsaSign 1.8: Sun RSA signature provider
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunEC 1.8: Sun Elliptic Curve provider (EC, ECDSA, ECDH)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunJSSE 1.8: Sun JSSE provider(PKCS12, SunX509/PKIX key/trust factories, SSLv3/TLSv1/TLSv1.1/TLSv1.2)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunJCE 1.8: SunJCE Provider (implements RSA, DES, Triple DES, AES, Blowfish, ARCFOUR, RC2, PBE, Diffie-Hellman, HMAC)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunJGSS 1.8: Sun (Kerberos v5, SPNEGO)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunSASL 1.8: Sun SASL provider(implements client mechanisms for: DIGEST-MD5, GSSAPI, EXTERNAL, PLAIN, CRAM-MD5, NTLM; server mechanisms for: DIGEST-MD5, GSSAPI, CRAM-MD5, NTLM)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: XMLDSig 1.8: XMLDSig (DOM XMLSignatureFactory; DOM KeyInfoFactory; C14N 1.0, C14N 1.1, Exclusive C14N, Base64, Enveloped, XPath, XPath2, XSLT TransformServices)
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: SunPCSC 1.8: Sun PC/SC provider
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.HttpClient <clinit>
DETALLADO: Apple 1.8: Apple Provider
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.useragent = Jakarta Commons-HttpClient/3.1
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.version = HTTP/1.1
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.connection-manager.class = class org.apache.commons.httpclient.SimpleHttpConnectionManager
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.cookie-policy = default
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.element-charset = US-ASCII
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.protocol.content-charset = ISO-8859-1
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.method.retry-handler = org.apache.commons.httpclient.DefaultHttpMethodRetryHandler@c18a75e
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.dateparser.patterns = [EEE, dd MMM yyyy HH:mm:ss zzz, EEEE, dd-MMM-yy HH:mm:ss zzz, EEE MMM d HH:mm:ss yyyy, EEE, dd-MMM-yyyy HH:mm:ss z, EEE, dd-MMM-yyyy HH-mm-ss z, EEE, dd MMM yy HH:mm:ss z, EEE dd-MMM-yyyy HH:mm:ss z, EEE dd MMM yyyy HH:mm:ss z, EEE dd-MMM-yyyy HH-mm-ss z, EEE dd-MMM-yy HH:mm:ss z, EEE dd MMM yy HH:mm:ss z, EEE,dd-MMM-yy HH:mm:ss z, EEE,dd-MMM-yyyy HH:mm:ss z, EEE, dd-MM-yyyy HH:mm:ss z]
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.connection.timeout = 60000
jul 02, 2017 8:12:49 PM org.apache.commons.httpclient.params.DefaultHttpParams setParameter
DETALLADO: Set parameter http.connection.timeout = 60000
[... cut ...]
Jun 27, 2017 10:06:19 PM org.apache.commons.httpclient.HttpMethodDirector executeWithRetry
FINEST: Attempt number 1 to process request
Jun 27, 2017 10:06:20 PM org.ros.internal.node.topic.DefaultPublisher onMasterRegistrationFailure
INFO: Publisher registration failed: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</Listener, http://ev3dev:44996/>, TopicIdentifier</rosout>>, Topic<TopicIdentifier</rosout>, TopicDescription<rosgraph_msgs/Log, acffd30cd6b6de30f120938c17c593fb>>>>
```

jabrena, 2017-07-02:
If you observe, in the middle of the first try, raise an exception, maybe exist a Threading problem. With faster computers never happen but with slow computer happen this threading problem.  I think that the brick connect but previously exist the Exception.

What is your opinion @adamantivm, @jubeira ?

To set the logs, it is easy:

``` 
# The following creates two handlers
handlers=java.util.logging.ConsoleHandler, java.util.logging.FileHandler
# Set the default logging level for the root logger
.level=ALL
# log level for the "com.example" package
org.ros.logging.level=ALL
# Set the default logging level
java.util.logging.ConsoleHandler.level=ALL
java.util.logging.FileHandler.level=ALL
# Set the default formatter
java.util.logging.ConsoleHandler.formatter=java.util.logging.SimpleFormatter
java.util.logging.FileHandler.formatter=java.util.logging.SimpleFormatter
# Specify the location and name of the log file
java.util.logging.FileHandler.pattern=/Users/jabrena/Documents/testOSX3.log
``` 

later, in the execution:

```
java -Djava.util.logging.config.file=/home/robot/log-config.properties -jar rosjava-helloworld-0.1.0-SNAPSHOT-all.jar 
``` 

I am going to see how to run the log configuration automatically not explicitly. :)

jubeira, 2017-09-27:
@jabrena did you find the solution to the problem? It's a bit hard to tell by looking at the logs from the outside; I couldn't reproduce an issue like this on my side; the XMLRPC timeouts are set to a couple of seconds, so it doesn't sound like a slow device could cause that.

jabrena, 2017-09-27:
Hi @jubeira, no I couldn't solve the problem. I did a lot of tests with the EV3 and I suspect that it is a threading problem. For Slow devices, some thread start before another, this is my explanation. With Raspberry Pi, I don't have that problem, but it is normal is a faster microprocessor. If you have time, we could try again. I have a stable solution using ROSBridge, but I will add specific support for raspberry Pi later, but I would like to publish nodes on EV3, but maybe I need your help or from someone for ROSJava team.

Juan Antonio

jubeira, 2017-09-27:
Interesting. It is true that rosjava uses threads and asynchronous tasks a lot, and you may be right it. Sorrry I cannot be more helpful on this one, the error message is a bit cryptic and I've never experienced this issue. Please let me know if you find something else.

jabrena, 2017-09-27:
This is my point of view from the beginning.
I think that this issue should be opened or create a new one to start again.
I could test with the devices but we should identify what is the right sequence of execution with independence of the speed of the device and later see how to solve the problem.

Juan Antonio