rosjava/rosjava_core #252: do not catkinmake

DavidHan008, 2017-06-30:
my ubutntu14.04 ros indigo
I am new learner.
I just follow the tutorial to build my packcage of java.
just follow
> cd src/rosjava_catkin_package_a
> catkin_create_rosjava_project rosjava_gradle_subproject_a
> cd ../..
> catkin_make
--------------------
but when i finally cakin_make
it have some problems:
[100%] Gradling tasks for rosjava_catkin_package_a
Downloading https://services.gradle.org/distributions/gradle-2.2.1-bin.zip

Exception in thread "main" javax.net.ssl.SSLException: java.security.ProviderException: java.security.InvalidKeyException: EC parameters error
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:208)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1914)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1872)
	at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1855)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1376)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1353)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1348)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:254)
	at org.gradle.wrapper.Download.downloadInternal(Download.java:56)
	at org.gradle.wrapper.Download.download(Download.java:42)
	at org.gradle.wrapper.Install$1.call(Install.java:57)
	at org.gradle.wrapper.Install$1.call(Install.java:44)
	at org.gradle.wrapper.ExclusiveFileAccessManager.access(ExclusiveFileAccessManager.java:65)
	at org.gradle.wrapper.Install.createDist(Install.java:44)
	at org.gradle.wrapper.WrapperExecutor.execute(WrapperExecutor.java:126)
	at org.gradle.wrapper.GradleWrapperMain.main(GradleWrapperMain.java:56)
Caused by: java.security.ProviderException: java.security.InvalidKeyException: EC parameters error
	at sun.security.pkcs11.P11Key$P11ECPublicKey.getEncodedInternal(P11Key.java:1024)
	at sun.security.pkcs11.P11Key.equals(P11Key.java:158)
	at java.util.ArrayList.indexOf(ArrayList.java:298)
	at java.util.ArrayList.contains(ArrayList.java:281)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:239)
	at sun.security.validator.Validator.validate(Validator.java:260)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:326)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:231)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:126)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1459)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:213)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:961)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:897)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1033)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1342)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1369)
	... 13 more
Caused by: java.security.InvalidKeyException: EC parameters error
	at sun.security.ec.ECParameters.getAlgorithmParameters(ECParameters.java:284)
	at sun.security.ec.ECPublicKeyImpl.<init>(ECPublicKeyImpl.java:59)
	at sun.security.pkcs11.P11Key$P11ECPublicKey.getEncodedInternal(P11Key.java:1021)
	... 28 more
Caused by: java.security.NoSuchProviderException: no such provider: SunEC
	at sun.security.jca.GetInstance.getService(GetInstance.java:83)
	at sun.security.jca.GetInstance.getInstance(GetInstance.java:206)
	at java.security.Security.getImpl(Security.java:697)
	at java.security.AlgorithmParameters.getInstance(AlgorithmParameters.java:199)
	at sun.security.ec.ECParameters.getAlgorithmParameters(ECParameters.java:279)
	... 30 more
make[2]: *** [rosjava_catkin_package_a/CMakeFiles/gradle-rosjava_catkin_package_a] Error 1
make[1]: *** [rosjava_catkin_package_a/CMakeFiles/gradle-rosjava_catkin_package_a.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
----------------------------------------
i don't know 
so please help me!

jubeira, 2017-06-30:
Hi @DavidHan008,
These packages require to download Gradle files to build. Your error sounds like a network issue between your computer and the Gradle server. Check if you can download those files at least manually to see if you find the problem.

DavidHan008, 2017-07-01:
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-2.14.1-all.zip
I solve it thx

DavidHan008, 2017-07-02:
@jubeira hi，another problem.
i want to catkin make rosjava_core
but have the same problems 
Downloading https://services.gradle.org/distributions/gradle-3.5.1-all.zip

Exception in thread "main" java.lang.RuntimeException: javax.net.ssl.SSLHandshakeException: Remote host closed connection during handshake
	at org.gradle.wrapper.ExclusiveFileAccessManager.access(ExclusiveFileAccessManager.java:78)
do you have some idea about download the zip and put it in where?

DavidHan008, 2017-07-02:
I can download this zip but I don't know how to manually to use it 
can you help me?

adamantivm, 2017-07-03:
@DavidHan008 apparently there is a regression on some versions of openjdk 8. See https://github.com/rosjava/rosjava_messages/issues/29

Perhaps you should try switching to another JDK and see if that fixes the issue?

ericson-cepeda, 2017-08-17:
To solve it in my case:

1. Download gradle and save it to: gradle/wrapper/.
2. set gradle-wrapper.properties to use the downloaded file.

```
wget https://services.gradle.org/distributions/gradle-4.1-bin.zip -O "$(pwd)/gradle/wrapper/gradle-4.1-bin.zip"

cat <<EOF > "$(pwd)/gradle/wrapper/gradle-wrapper.properties"
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists
distributionUrl=gradle-4.1-bin.zip
EOF
```

stevensu1838, 2017-11-10:
Hi David, I am confronting the same issue. How did you get these following lines:
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists
distributionUrl=https://services.gradle.org/distributions/gradle-2.14.1-all.zip

jubeira, 2017-11-22:
We just found another workaround for this.
OpenJDK-8 or Oracle's JDK 8 doesn't have this issue. Installing one of those and setting `JAVA_HOME` environment variable to use it (e.g. `export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64`) should solve this issue.