rosjava/rosjava_core #253: Possible to generate bytecode for older java version?

broesdecat, 2017-08-09:
Hi,

I am trying to generate java code for ros messages for Kuka Sunrise.
The basic setup works fine, except that Kuka Sunrise runs on Java 6, and my catkin workspace on docker runs on Java 8. Hence, generated java code for ros messages does not work on the robot (unsupportedClassVersionError). Due to various other issues, I cannot downgrade my container to a lower Java version (Gradle having issues with openjdk, oracle java<1.8 installers being no longer available).
So I was wondering whether rosjava / genjava support cross-compiling from Java 8 to Java 6, as that would also resolve my problem.

Thx!
Broes

jubeira, 2017-08-09:
Hi @broesdecat!
Yes, it is possible indeed. You will have to compile `rosjava_core` again, adding 
```
sourceCompatibility = 1.6
targetCompatibility = 1.6
```
to the `build.gradle` file inside `subprojects` task.
I did that myself and uploaded the resulting artifacts to a personal mvn repo here: https://github.com/jubeira/rosjava_mvn_repo/commits/java_6_improvements.

If you compile the code on your own, you can verify if the bytecodes are OK by extracting a class from a Jar, and then running
`javap -v extractedClass.class | grep major`

Hope it helps!

broesdecat, 2017-08-10:
@jubeira Great answer, worked exactly as advertised and does the job :) Thx!

jubeira, 2017-08-10:
glad it helped :smile: 
could you close this issue please?

closed 2017-08-10 by broesdecat