rosjava/rosjava_core #254: ROSjava and android ROS commands

dkati, 2017-08-31:
Good evening people :) 
i am trying to understand if it is possible to "send" a ROS command like 'rostopic list' to a connected PC via android device.

i have fairly understand how the ROS apks works,i made one on my own too but i cant find a way to actually read the topics of the PC i am connected to,from my apk.Any ideas ? do i need rosjava ?thanks

jubeira, 2017-08-31:
Hi @dkati,
What commands do you actually need? Do you want just the list of topics, or subscribing / publishing to one of them?

Personally, I never did something like sending a `rostopic list` from rosjava, but it should be possible to do so.

dkati, 2017-08-31:
exactly.i want to list the topics of the ros PC i am connected to....
the same way i do it from PC ,i want to do it via android device.

my app allows to subscribe to custom topic names.so if the user is away from PC ,he has to know what topics trully exist to ros master.so he have to somehow see the 

> rostopic list 

output

jubeira, 2017-08-31:
I'd take a look around here: https://github.com/rosjava/rosjava_core/blob/kinetic/rosjava/src/main/java/org/ros/master/client/MasterStateClient.java

Unfortunately, it seems that this operation in particular hasn't been implemented: https://github.com/rosjava/rosjava_core/blob/kinetic/rosjava/src/main/java/org/ros/master/client/MasterStateClient.java#L84. I'm not 100% sure, but at first sight it seems that is what you are looking for.

As far as I know rostopic is implemented in Python, so you can't use that code on Android. If you really need this feature, I'd suggest you to try to implement that operation. If you send a PR with that fix we can  make a new release to the maven repo. I offer myself to review and test this if you choose to go that way.

dkati, 2017-08-31:
thats seems really DEEP for me mister :P ok..thanks for your time!!thanks for all!

jubeira, 2017-08-31:
haha yeah, it looks a bit complex at first sight, but perhaps it's not that terrible. Probably all the functionality needed to implement this is somewhere around rosjava, it's a matter of putting the pieces together.
Sorry I cannot be more helpful for now.. If more people need this feature perhaps it is worth investing some time on this.

willchamberlain, 2017-09-01:
For anyone venturing into this, the python implementation is in https://github.com/ros/ros_comm/blob/lunar-devel/tools/rostopic/src/rostopic/__init__.py , and `rostopic list` looks to be implemented on line 1187 as `def _rostopic_list(....)....` . Have fun.

dkati, 2017-09-01:
can u give me more instruction mister :) ?

willchamberlain, 2017-09-04:
Not really : I haven't ever used it, I just have the source code checked out in case I ever needed to.

willchamberlain, 2017-09-04:
Bit of googling found
- this, on hitting the ROS Master with XMLRPC : http://wiki.ros.org/ROS/Master_API
- these on Java XMLRPC: http://tldp.org/HOWTO/XML-RPC-HOWTO/xmlrpc-howto-java.html. ;  https://github.com/gturri/aXMLRPC
which might give you a starting point to recreate the rostopic functions in an Android app

orat, 2019-01-30:
Any news about the functionality of rostopic in java? If it is not implemented yet, maybe I can do it. Can sombody can give me some hints?

jubeira, 2019-01-30:
Hi @orat,
Please check this comment; that's what I found about the subject back then: https://github.com/rosjava/rosjava_core/issues/254#issuecomment-326287652.

Note that there's also [this post](https://discourse.ros.org/t/android-app-for-ros-inspection/3100/14) on ROS discourse; someone already made an app with this, and it's open source: https://github.com/joostvanstuijvenberg/IntROSpect (it's not using `rosjava`).

If you ask me, I would try to implement it as part of Rosjava; it would be nice to add something like this to  `rosjava_extras` or `rosjava_helpers`.

jubeira, 2019-01-30:
Yes, that would work as well.

On second thought, perhaps a good idea would be to mimic ROS CLI tools: https://github.com/ros/ros_comm/blob/melodic-devel/tools/rostopic/src/rostopic/__init__.py#L1154-L1176. Just as an example, `get_topic_list` uses `rosgraph.Master` under the hood to get the system state, and then it filters out the relevant information. Maybe implementing an analogue on rosjava is possible.

orat, 2019-02-01:
ok, I will think about to additionally implement a version corresponding to ROS CLI tools. At the moment i have simply implemented the missing method in org.ros.master.client.MasterStateClient:

 public List<TopicDeclaration> getPublishedTopics(String subgraph) {
    // TODO(keith): Figure out what to turn the topic definition into.
   Response<List<TopicDeclaration>> result = masterClient.getPublishedTopics(caller.getName(), subgraph);
    return result.getResult();
  }

In the original code there is the comment from "keith". Do you know what it means?

orat, 2019-02-01:
Why is the argument "caller" of the medhods of the class org.ros.master.client.MasterStateClient of type Node and not String. In the current implementation only a name as a String is used (caller.getName()? If the the argument "Node caller" make sense, what is the node I should use?

jubeira, 2019-02-01:
> In the original code there is the comment from "keith". Do you know what it means?

Sorry, I'm out of context here.

> Why is the argument "caller" of the medhods of the class org.ros.master.client.MasterStateClient of type Node and not String. In the current implementation only a name as a String is used (caller.getName()? If the the argument "Node caller" make sense, what is the node I should use?

I honestly don't know either, but my first guess would be that the node names are actually not plain strings, but graph names with certain constraints. While a string would do, I think it's a matter of being consistent. I'd look somewhere else where `MasterStateClient` is used and try to use it the same way.

orat, 2019-02-04:
Thanks for your answer. Yes your are right,  GraphNode and not String is used and makes sense for me. I looks a bit strange for me not use GraphNode directly (encapsulated into Node). At the moment this is not important.

I implemented the getPublishedTopics() as described in this thread and run into further problems.  A null pointer exception is thrown, which has its origin in the class org.ros.internal.response.TopicListResultFactory.
In line 45 of this class there is the line "new TopicDescription(type, null null)" which looks not correct. Instead of the null arguments  a definition of type String and MD5 checksum has to be used. I have no idea what to do. 
 
I have asked the author of the class but he can not help me. 

Anybody else?

At this point I have the feeling that it cauld be too difficult to complete the code and fix the issues. Maybe the author of the non-rosjava INTRosSpec code runs into the same problems and has descided to implement the functionality without the usage of rosjava.

jubeira, 2019-02-07:
> I implemented the getPublishedTopics() as described in this thread and run into further problems. A null pointer exception is thrown, which has its origin in the class [...]

I'm very sorry; I'm not into those details right now. I've seen MD5 checksums somewhere else in the code; if I was trying to implement this I'd try to look somewhere else the way they are used and what they are used for. Sounds like you need the name and the checksum of the topics you are listing, but I'm out of context. I'm a bit packed right now to look into this; if I free some of my time I'll try to take a look.

orat, 2019-02-08:
Thanks again for your time. It motivates me not to give up:-) I think I can spend next week some time trying to understand what values the missing parameters should have.

orat, 2019-02-08:
Surprisingly I think I have found what to do:  In org.ros.internal.node.response.TopicListResultFactory.java line 45 "new TopicDescription(type, null, null)" has to be substituted by "org.ros.internal.message.topic.TopicDescripition.newFromType(type)" I can verify it next week. Should I than create a pull-request, if it works?

talregev, 2019-06-16:
You should create a PR, and please do :)

orat, 2019-07-19:
Hi, 
I am back to rosjava and I have it on my todo list. Is any tutorial available to create a pullrequest for the rosjava sources or any hints what I have to consider?

best regards
Oliver

FunkyMax, 2019-11-25:
To achieve something like `rostopic list` one only needs these three lines of code. It's written in Kotlin but one should be able to convert it into java code.

```
val masterClient = MasterClient(masterUri)
val systemState = masterClient.getSystemState(GraphName.of("WHATEVER"))
val topicList = systemState.result.topics
```

Just iterate through topic list and you'll be able to retrieve all topics with corresponding publishers and subscribers.