rosjava/rosjava_core #255: Can't bind hostname to ip correctly!

erow, 2017-09-09:
My system environment is Jessie on raspberry pi. 
I create a subscriber at '/card'.And there's no way to publish any topic.
I got an error message:

> Sep 09, 2017 7:49:59 AM org.ros.internal.node.topic.DefaultSubscriber$1 onMasterRegistrationSuccess
> INFO: Subscriber registered: Subscriber<Topic<TopicIdentifier</card>, TopicDescription<card_reader/card, c52852a769436958171eee646747bd88>>>
> 
> Sep 09, 2017 7:50:29 AM org.ros.internal.node.topic.UpdatePublisherRunnable run
> SEVERE: org.ros.exception.RosRuntimeException: Connection exception: raspberrypi/218.205.57.154:45771
> Sep 09, 2017 7:50:29 AM org.ros.internal.transport.ConnectionTrackingHandler exceptionCaught
> SEVERE: Channel exception: java.net.ConnectException: connection timed out

It's strange **raspberrypi/218.205.57.154** ,it should be '127.0.0.1'. 

> 127.0.0.1	raspberrypi
> 192.168.1.25   DESKTOP-EROW `

 And I have no idea to the problem.

jubeira, 2017-09-11:
Hi @erow,
How are you executing your nodes? Where is your master running, and how are you initializing the nodes with their `NodeConfiguration`? That IP address sounds like an external one, not a local one (they start in 10.xxx, 192.168.xxx, 172.16.xx or 169.254.xx.

You should be doing something like this to start the node: https://github.com/ekumenlabs/tangobot/blob/master/tangobot_app/app/src/main/java/com/ekumen/tangobot/application/MainActivity.java#L398.

erow, 2017-09-12:
I diidn't change my `NodeConfiguration`. But my code works on my x86 computer.I build them on my PC, and copy the `build` directory to Raspberry pi.It did go wrong just on pi.

jubeira, 2017-09-12:
can you copy the code you are using to execute the node? What does your `/etc/hosts` file look like?

erow, 2017-09-13:
I copy the code from `ros.RosRun`
```
if (argv.length == 0) {
            printUsage();
            System.exit(1);
        }

        CommandLineLoader loader = new CommandLineLoader(Lists.newArrayList(argv));
        String nodeClassName = loader.getNodeClassName();
        System.out.println("Loading node class: " + loader.getNodeClassName());
        NodeConfiguration nodeConfiguration = loader.build();
        nodeConfiguration.setMasterUri(URI.create(MasterUri));
        NodeMain nodeMain = null;

        try {
            nodeMain = loader.loadClass(nodeClassName);
        } catch (ClassNotFoundException var6) {
            throw new RosRuntimeException("Unable to locate node: " + nodeClassName, var6);
        } catch (InstantiationException var7) {
            throw new RosRuntimeException("Unable to instantiate node: " + nodeClassName, var7);
        } catch (IllegalAccessException var8) {
            throw new RosRuntimeException("Unable to instantiate node: " + nodeClassName, var8);
        }

        Preconditions.checkState(nodeMain != null);
        NodeMainExecutor nodeMainExecutor = DefaultNodeMainExecutor.newDefault();
        nodeMainExecutor.execute(nodeMain, nodeConfiguration);
```
My hosts:

```
127.0.1.1	raspberrypi
192.168.1.25    DESKTOP-RGJ3O7D
```
By the way I always got error when I compiled rosjava.

> Exception in thread "main" javax.net.ssl.SSLException: java.security.ProviderException: java.security.InvalidKeyException: EC parameters error

jubeira, 2017-09-13:
Ok, but where are you getting the MasterUri from?
```
nodeConfiguration.setMasterUri(URI.create(MasterUri));
```
Perhaps you are trying to target your external IP address somehow, and that's why you are getting the exception. The hosts file looks OK.

erow, 2017-09-29:
I still don't know why. But I find it's more reliable to set `ROS_IP`.

jubeira, 2017-09-29:
If I had to debug this, I would check the following items:
- What is the `MasterUri` that you are setting/ creating.
- Once you start your system, check the info using `rostopic info` of the topics that you are using. This will tell you which nodes are subscribing/ publishing to those topics.
- The node information provided by `rostopic info` will tell how the nodes were registered (i.e. if they are using the system's hostname or IP address). Then, that will let you know if a node A can subscribe to topic B, that has node C as a subscriber.

For example, if I publish a message to the topic `/foo`, and I subscribe from the terminal, I will see something like this:
```
juan@juan-HOTH:~$ rostopic info /foo
Type: std_msgs/String

Publishers: 
 * /rostopic_12101_1506689204842 (http://juan-HOTH:34297/)

Subscribers: 
 * /rostopic_12112_1506689211718 (http://juan-HOTH:42517/)
```
This means that the node ending in `842` is publishing and is advertised using the `HOSTNAME`, and the subscriber (ending in `718`) is subscribing and also using the hostname. For the system to work properly, the machine running the subscriber has to be able to resolve the hostname (in this case, the IP doesn't matter).

On the other side, if I first import `ROS_IP` before publishing the message, this is what I see:
```
juan@juan-HOTH:~$ rostopic info /foo
Type: std_msgs/String

Publishers: 
 * /rostopic_12194_1506689376538 (http://10.34.0.107:41761/)

Subscribers: 
 * /rostopic_12206_1506689378519 (http://juan-HOTH:36039/)
```
As you can see, now the publisher is using its IP address, not the hostname. So in this case, the subscriber will need to resolve that IP address, and won't care about the hostname.

Does that help...?

jubeira, 2017-09-29:
Hmm interesting...
Anyway, I'm glad that it works now! Should we keep this issue open or can we close it?

erow, 2017-09-30:
I think I won't have this problem again, so close it maybe better.

closed 2017-11-13 by jubeira