rosjava/rosjava_core #256: How to choose a right maven repository url?

erow, 2017-10-09:
I find the default url is very slow when compiling the project, It likes to get stuck as follows. I guess it is because the repository url is slow, and I want to change to a different one for try.
```
####
#### Running command: "make cmake_check_build_system" in "/home/wasai/ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/wasai/ws/build/rosjava_client"
####
[100%] 
Gradling tasks for rosjava_client
...........................get stuck here......................................
```

jubeira, 2017-10-09:
Hi @erow,
Perhaps https://answers.ros.org/ is a better place for these questions, as this is a regular Gradle problem a not a rosjava one.
The repository URL is [rosjava_mvn_repo](https://github.com/rosjava/rosjava_mvn_repo). Most of the dependencies that you use in a rosjava project are there, so you would need to get those artifacts somewhere else in first place. Does that make sense?

To add a new URL, simply add the following lines to your Gradle buildscript:
```
repositories {
    maven {
        url "http://your-url.com"
    }
}
```
More detail here: https://stackoverflow.com/questions/30114860/multiple-maven-repositories-in-one-gradle-file

erow, 2017-10-10:
Sorry.  It's a new fish to use java. And I was looking for ".m2\Repository". I didn't think about gradle to change it. But now I meet a big trouble. I will open a new issue soon.

drigz, 2018-02-26:
@erow, did you make any progress on this issue?

erow, 2018-05-05:
I don't know. I just reinstalled and it worked.

closed 2018-05-05 by erow