rosjava/rosjava_core #257: Can not subscribe topics correctly.

erow, 2017-10-10:
```
---------------------------------roswtf----------------------------------------
WARNING The following node subscriptions are unconnected:
 * /audio:
   * /audio/stop
 * /serial_node:
   * /chassis/led
   * /chassis/magnetic/init

Found 1 error(s).

ERROR The following nodes should be connected but aren't:
 * /serial_node->/rosjava_client (/chassis/card)
 * /serial_node->/rosjava_client (/chassis/ir)
---------------------------------rosnode info----------------------------------------
Node [/rosjava_client]
Publications: 
 * /chassis/changeMode [std_msgs/String]
 * /audio/play [std_msgs/String]
 * /rosout [rosgraph_msgs/Log]
 * /chassis/touch [std_msgs/Empty]
 * /chassis/cmd [std_msgs/String]
 * /chassis/param [card_reader/param]
 * /chassis/moveT [card_reader/move]

Subscriptions: 
 * /chassis/ir [std_msgs/Bool]
 * /chassis/card [std_msgs/String]
 * /chassis/touch [std_msgs/Empty]

Services: None

contacting node http://127.0.0.1:43343/ ...
Pid: 1215
Connections:
 * topic: /chassis/touch
    * to: /rosjava_client
    * direction: outbound
    * transport: TCPROS
 * topic: /chassis/moveT
    * to: /serial_node
    * direction: outbound
    * transport: TCPROS
 * topic: /audio/play
    * to: /audio
    * direction: outbound
    * transport: TCPROS
 * topic: /chassis/param
    * to: /serial_node
    * direction: outbound
    * transport: TCPROS
 * topic: /chassis/changeMode
    * to: /serial_node
    * direction: outbound
    * transport: TCPROS
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /chassis/cmd
    * to: /serial_node
    * direction: outbound
    * transport: TCPROS
 * topic: /chassis/touch
    * to: /rosjava_client (http://127.0.0.1:43343/)
    * direction: inbound
    * transport: TCPROS
```
rosjava can not recv the message, but it's fine to `echo`:
```
rostopic echo /chassis/card
data: 0003935828
---
```
And the Log files show the problem:
```
19903 [pool-1-thread-37] ERROR org.ros.internal.node.topic.UpdatePublisherRunnable  - org.ros.exception.RosRuntimeException: Connection exception: raspberrypi/218.205.57.154:44945
19920 [New I/O client boss #1-0] ERROR org.ros.internal.transport.ConnectionTrackingHandler  - Channel exception: java.net.ConnectException: connection timed out
21503 [pool-1-thread-56] ERROR org.ros.internal.node.topic.UpdatePublisherRunnable  - org.ros.exception.RosRuntimeException: Connection exception: raspberrypi/218.205.57.154:44945
21511 [New I/O client boss #3-0] ERROR org.ros.internal.transport.ConnectionTrackingHandler  - Channel exception: java.net.ConnectException: connection timed out
38028 [pool-1-thread-52] ERROR org.ros.internal.node.topic.UpdatePublisherRunnable  - org.ros.exception.RosRuntimeException: Connection exception: raspberrypi/218.205.57.154:34259
38032 [New I/O client boss #4-0] ERROR org.ros.internal.transport.ConnectionTrackingHandler  - Channel exception: java.net.ConnectException: connection timed out
```

jubeira, 2017-10-10:
Hi @erow,
Can you be more specific? What are you trying to do? Where are you running the nodes that should be connected with each other? Can you show me how you are executing rosjava nodes?

At a quick glance, this seems to be the classical networking issue... Have you taken a look at http://wiki.ros.org/ROS/Technical%20Overview#Establishing_a_topic_connection? If you can echo the topic from somewhere, it means that the node is running, but perhaps your subscriber can't resolve the publisher's address.
In your logs I see a mix of public and private IP addresses. Try using `rostopic info` for each of your topics, that will give us a hint.

erow, 2017-10-11:
I just publish a message by:`rostopic pub /chassis/card std_msgs/String "data: 'asd'" --once`
But rosjava can not  receive it. And I can get it by`rostopic echo /chassis/card`.
And it's strange that rosjava can publish normally, and the topic `/chassis/touch`  works well.
I print some debug log:
```
1046745 [Thread-10] DEBUG org.apache.commons.httpclient.HttpMethodBase  - Resorting to protocol version default close connection policy
1046745 [Thread-10] DEBUG org.apache.commons.httpclient.HttpMethodBase  - Should close connection, using HTTP/1.0
1046745 [Thread-10] DEBUG org.apache.commons.httpclient.HttpConnection  - Releasing connection back to connection manager.
1049688 [XML-RPC-0] DEBUG org.apache.xmlrpc.server.XmlRpcStreamServer  - execute: ->
1049699 [XML-RPC-0] DEBUG org.apache.xmlrpc.server.XmlRpcStreamServer  - execute: Request performed successfully
1049700 [XML-RPC-0] DEBUG org.apache.xmlrpc.server.XmlRpcStreamServer  - execute: <-
1058078 [pool-1-thread-25] ERROR org.ros.internal.node.topic.UpdatePublisherRunnable  - org.ros.exception.RosRuntimeException: Connection exception: raspberrypi/218.205.57.154:36337
1058082 [New I/O client boss #3-0] ERROR org.ros.internal.transport.ConnectionTrackingHandler  - Channel exception: java.net.ConnectException: connection timed out
```
And I use tcpdump to capture the request:
```
        POST /RPC2 HTTP/1.1
        Host: localhost:11311
        Accept-Encoding: gzip
        User-Agent: xmlrpclib.py/1.0.1 (by www.pythonware.com)
        Content-Type: text/xml
        Content-Length: 393

<methodCall>
    <methodName>registerPublisher</methodName>
    <params>
        <param>
            <value><string>/rostopic_11910_1507690526359</string></value>
        </param>
        <param>
            <value><string>/chassis/card</string></value>
        </param>
        <param>
            <value><string>std_msgs/String</string></value>
        </param>
        <param>
            <value><string>http://raspberrypi:36147/</string></value>
        </param>
    </params>
</methodCall>
        HTTP/1.0 200 OK
        Server: BaseHTTP/0.3 Python/2.7.13
        Date: Wed, 11 Oct 2017 02:55:26 GMT
        Content-type: text/xml
        Content-length: 374

<methodResponse>
<params>
    <param>
        <value><array><data>
            <value><int>1</int></value>
            <value><string>Registered [/rostopic_11910_1507690526359] as publisher of [/chassis/card]</string></value>
            <value><array><data>
                <value><string>http://127.0.0.1:35593/</string></value>
            </data></array></value>
        </data></array></value>
    </param>
</params>
</methodResponse>
      
        POST / HTTP/1.1
        Host: 127.0.0.1:35593
        Accept-Encoding: gzip
        User-Agent: xmlrpclib.py/1.0.1 (by www.pythonware.com)
        Content-Type: text/xml
        Content-Length: 349

<methodCall>
<methodName>publisherUpdate</methodName>
<params>
[... 118 lines cut ...]
                <value><array><data>
                    <value><int>1</int></value>
                    <value><string>Unregistered [/rostopic_11910_1507690526359] as provider of [/chassis/card]</string></value>
                    <value><int>1</int></value>
                </data></array></value>
            </data></array></value>
            <value><array><data>
                <value><array><data>
                    <value><int>1</int></value>
                    <value><string>Unregistered [/rostopic_11910_1507690526359] as provider of [/rostopic_11910_1507690526359/get_loggers]</string></value>
                    <value><int>1</int></value>
                </data></array></value>
            </data></array></value>
            <value><array><data>
                <value><array><data>
                    <value><int>1</int></value>
                    <value><string>Unregistered [/rostopic_11910_1507690526359] as provider of [/rostopic_11910_1507690526359/set_logger_level]</string></value>
                    <value><int>1</int></value>
                </data></array></value>
            </data></array></value>
        </data></array></value>
    </param>
</params>
</methodResponse>
```

erow, 2017-10-20:
I found what the problem is.
And I have a temporary solution.

```java
package org.ros.address;
public static InetAddress newFromHostString(String host) {
        try {
            if (InetAddresses.isInetAddress(host)) {
                return InetAddress.getByAddress(host, InetAddresses.forString(host).getAddress());
            }

            if (host.equals("localhost")) {
                return InetAddress.getByAddress("localhost", InetAddresses.forString("127.0.0.1").getAddress());
            }
        } catch (UnknownHostException var4) {
            throw new RosRuntimeException(var4);
        }

        Collection<InetAddress> allAddressesByName = getAllInetAddressesByName(host);
       ...
        return address;
    }
```
` getAllInetAddressesByName(host);` cann't get the right ip address.
So I change my hostname to localhost, and it works.

jubeira, 2017-10-20:
@erow that's why I was asking how you were executing the node. This is a network / Java problem; whenever you have issues with this always print the host that you are using to execute the node, and use `rostopic info` on the topics that are not working well. That will give you good hints and you won't need to deal with `tcpdump`, which gives you too much low level info that you probably don't need.

Again, I suggest you to read http://wiki.ros.org/ROS/Technical%20Overview#Establishing_a_topic_connection; this applies to ROS and rosjava as well.

Can you please close this issue then?

erow, 2017-10-20:
I still don't know why `Adress.getAllByName(host)` fail, and `InetAddress.getByName` work. It seems the problem of dnsjava. What's the difference of them? And why not just use  `InetAddress.getByName`. It works under both of my machines.

jubeira, 2017-10-20:
I _think_ that is related with your network setup. At some point, your system will need to resolve the IP address against the hostname that you provide. 

For example, the resolution for `localhost` in a Linux system is most likely in a text file `/etc/hosts` that binds that hostname to the loopback IP address. For other hostnames, you will need to either add another mapping to that `hosts` file, or set up a DNS server that can resolve the hostname that you want to an IP address.

In any case, the underlying issue is how to resolve the IP address of a given hostname.

closed 2017-11-07 by jubeira