rosjava/rosjava_core #268: testRegisterPublisher and testRegisterPublisherRetries are broken

drigz, 2018-02-26:
aa381199455379d82d2bc5814933ceeded2ee793 (#238) has broken testRegisterPublisher and testRegisterPublisherRetries. The stacktrace in both cases is similar:

```
java.lang.AssertionError: 
	at org.junit.Assert.fail(Assert.java:91)
	at org.junit.Assert.assertTrue(Assert.java:43)
	at org.junit.Assert.assertTrue(Assert.java:54)
	at org.ros.internal.node.MasterRegistrationTest.testRegisterPublisher(MasterRegistrationTest.java:45)
```

The [assertion](https://github.com/rosjava/rosjava_core/blob/fe2b833e3f08bf00662e80eb6737cc7824c975ac/rosjava/src/test/java/org/ros/internal/node/MasterRegistrationTest.java#L45) on line 45 is:

```
assertTrue(publisherListener.awaitMasterUnregistrationSuccess(1, TimeUnit.SECONDS));
```

I am testing with `./gradlew test` on Debian Buster with Gradle 2.14.1 and Oracle Java 1.8.0.

@stratomda Any idea what the right fix here is? Do the tests fail for you as well?

dambrosio, 2018-02-26:
I will need to get my setup configured on my personal computer and then run the tests. Unfortunately, I did not run the gradle tests after my commit (☹️ ) so I cannot say if the tests failed or passed for me. I will do this today and try and get back with you guys.

PS. I am using my personal account instead of my work. Just a heads up.

drigz, 2018-02-26:
Thanks! Let us know if you make any progress!

dambrosio, 2018-02-27:
@drigz It seems as if the Master is successfully sending the unregisterPublisher xml response by looking at the network traffic. I changed the publisher to /register so I could filter the network traffic.

Here is the unregisterPublisher request triggered by `publisher.shutdown()` on line 44 of MasterRegistrationTest.java:
```
Transmission Control Protocol, Src Port: 59259, Dst Port: 38705, Seq: 1, Ack: 1, Len: 424
Hypertext Transfer Protocol
eXtensible Markup Language
    <?xml
        version="1.0"
        encoding="UTF-8"
        ?>
    <methodCall>
        <methodName>
            unregisterPublisher
            </methodName>
        <params>
            <param>
                <value>
                    /node
                    </value>
                </param>
            <param>
                <value>
                    /register
                    </value>
                </param>
            <param>
                <value>
                    http://127.0.0.1:35013/
                    </value>
                </param>
            </params>
        </methodCall>

```
And here is the response from the Master:
```
Transmission Control Protocol, Src Port: 38705, Dst Port: 59259, Seq: 1, Ack: 425, Len: 329
Hypertext Transfer Protocol
eXtensible Markup Language
    <?xml
        version="1.0"
        encoding="UTF-8"
        ?>
    <methodResponse>
        <params>
            <param>
                <value>
                    <array>
                        <data>
                            <value>
                                <i4>
                                    1
                                    </i4>
                                </value>
                            <value>
                                Success
                                </value>
                            <value>
                                <i4>
                                    1
                                    </i4>
                                </value>
                            </data>
                        </array>
                    </value>
                </param>
            </params>
        </methodResponse>
```

I think what is happening is the ListenerGroup in the publisher is being shutdown before the master unregister signal is caught and thus causing the timeout. It is almost like there should be a check to see if the `signalOnMasterUnregistrationSuccess()` function has been executed before the `listeners.shutdown()` is called.

File: _DefaultPublisher.java_
```java
  @Override
  public void shutdown(long timeout, TimeUnit unit) {
    signalOnShutdown(timeout, unit);
    outgoingMessageQueue.shutdown();
    // Add a timer check to see if we got the unregister signal from master, then proceed to shutdown all listeners 
    listeners.shutdown();
  }
```
Here is where the unregister gets triggered in the DefaultPublisher.java class

File: _DefaultPublisher.java_
```java
@Override
  public void signalOnMasterUnregistrationSuccess() {
    final Publisher<T> publisher = this;
    listeners.signal(new SignalRunnable<PublisherListener<T>>() {
      @Override
      public void run(PublisherListener<T> listener) {
        listener.onMasterUnregistrationSuccess(publisher);
      }
    });
  }
```
Any ideas?
@jubeira adding you to this for input, let me know if you have questions.

PR #297 (2019-04-10): Fix for failing tests in class MasterRegistrationTest.
Waits for all signals to be delivered before shutting down listeners.
Creates new testUnregisterPublisherFailure() test in class MasterRegistrationTest.
closes #268
closed 2019-04-10 by jubeira