rosjava/rosjava_core #272: No error for ServiceServer on duplicate topic

broesdecat, 2018-03-23:
Hi,

I observed the following:
- a serviceserver A running on servicename X, which is properly receiving calls from a serviceclient
- another serviceserver B is started, accidentally also on X
- logging info reports initialization of B as succesful, neither does any other node complain
- messages sent by the client keep arriving only at A

Is this the expected behavior?

Thanks!
Broes

jubeira, 2018-03-26:
Hi @broesdecat,

Thanks for the observation. It doesn't look like expected behavior to me; at least I would expect some sort of warning/ error message. Ideally, the behavior should be the same as in `rospy` or `roscpp` (I would start by checking that).

If this is a problem for you, feel free to open a PR with a proposed fix. Otherwise, I will take a look when I have some time (it might take a bit long).

SpyrosKou, 2022-06-18:
Hi, 
I would like to add some information on this issue.

When the two service servers A and B that have the same service name X are started on the same Node the issue does not occur. I only occurs when service servers A and B are started on different Nodes.
When the two service servers are started on the same Node a DuplicateServiceException is thrown. 

A good behaviour could be to throw the DuplicateServiceException even if a Service Server with the same service name already is registered but on a different Node.

More over, the existence of two service servers A and B using the same service name X on the same Node is tested in  org.ros.node.service.ServiceIntegrationTest.