rosjava/rosjava_core #279: NullPointerException during node shutdown

qswawrq, 2018-05-21:
This is a question about #27. I saw the same exception during some normal shutdowns. Do we have a fix for that? Or we should simply ignore it and avoid catch this type of Exception during shutdown.
```
org.ros.exception.RosRuntimeException: java.lang.NullPointerException
	at org.ros.address.AdvertiseAddress.toString(AdvertiseAddress.java:123)
	at java.lang.String.valueOf(String.java:2994)
	at java.lang.StringBuilder.append(StringBuilder.java:131)
	at org.ros.address.AdvertiseAddress.toUri(AdvertiseAddress.java:109)
	at org.ros.internal.node.service.DefaultServiceServer.getUri(DefaultServiceServer.java:117)
	at org.ros.internal.node.client.MasterClient.unregisterService(MasterClient.java:91)
	at org.ros.internal.node.DefaultNode.shutdown(DefaultNode.java:424)
        ...
        at org.ros.internal.transport.queue.MessageDispatcher$1.run(MessageDispatcher.java:125)
	at org.ros.internal.transport.queue.MessageDispatcher$1.run(MessageDispatcher.java:122)
	at org.ros.concurrent.EventDispatcher.loop(EventDispatcher.java:43)
	at org.ros.concurrent.CancellableLoop.run(CancellableLoop.java:56)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at org.ros.internal.transport.tcp.TcpRosServer$1.call(TcpRosServer.java:91)
	at org.ros.internal.transport.tcp.TcpRosServer$1.call(TcpRosServer.java:88)
	at org.ros.address.AdvertiseAddress.toString(AdvertiseAddress.java:121)
	... 17 more
```

jubeira, 2018-05-21:
Hi @qswawrq, thanks for reporting this. It seems that this issue was closed long ago, but there's no reference to the fix.
Does this happen always to you, or in some particular situation? Do you have some simple code sample to reproduce? Does it have any functional consequences in your code?

qswawrq, 2018-05-21:
Thanks for reply so quickly. I think #27 provides a good way to reproduce it. Based on my understanding, SlaveServer shutdown will cause its TcpRosServer's outgoingChannel to be null. In that case, calling ```unregisterService``` in the for loop will trigger a NPE in advertiseAddress.toUri(), because advertiseAddress's portCallable require a non-null outgoingChannel. It is not clear to me why it is not happened all the time. The only functional consequences I can think of are some residual rosjava processes after shutdown.

jubeira, 2018-05-22:
Alright, thanks for the info. I've been kinda short on time lately, but I will see what I can do.

If you have a proposed fix for this, please feel free to open a PR; I will review it ASAP if you do.

qswawrq, 2018-05-22:
Sounds good. Will do.

qswawrq, 2018-05-22:
I just realized that it might not be a simple fix. Because my fix is to revert #242, which seems to cause other issues.
Note: I just confirm it only happens in normal shutdown. It will not be triggered when you exit by Ctrl+C.