rosjava/rosjava_core #280: ROSJava Catkin package support for ROS Lunar o ROS Melodic

jabrena, 2018-05-26:
Hi,

I would like to know if the tools to create a Catkin package for ROS Java has support for ROS Lunar and ROS Melodic.
http://wiki.ros.org/melodic/Installation

Many thanks in advance

Cheers

Juan Antonio

jubeira, 2019-03-18:
@rjcausarano I'd like to move forward with this, in particular for ROS Melodic.
Can you please follow the installation from source tutorials in Ubuntu 18 to check if they work, and point out issues if they appear?
- This is the [entrypoint](http://wiki.ros.org/rosjava) for rosjava tutorials in general.
  - This is the tutorial to install [rosjava from source](http://wiki.ros.org/rosjava/Tutorials/kinetic/Source%20Installation). 

Note that you will need ROS M previously installed, and replace `kinetic` with `melodic` when needed when following the tutorial. Thanks in advance!

bbferka, 2019-04-03:
@jubeira Hi, I have rosjava on melodic from source. I don't have time to manually test the tutorials, but I could run some unit tests if there are any?

jubeira, 2019-04-03:
Hi @bbferka,
The unit tests can be run from `rosjava_core` using `./gradlew test` task. I don't think there will be a difference on that end.
However, we noticed two things:
- Some packages listed in `rosjava_messages` package are not listed in kinetic. Those should be removed from the dependencies until they are released. In particular [this tutorial](http://wiki.ros.org/rosjava/Tutorials/kinetic/Source%20Installation) should fail when installing dependencies using `rosdep`.
- Some message definitions in `rosjava_mvn_repo` need to be updated because they are not present in kinetic. This will need a separate branch there as well to be able to maintain the two distributions.