rosjava/rosjava_core #288: rosjava builds failing on Kinetic

tfoote, 2019-01-22:
There is a regression in the kinetic builds of rosjava_core and a few other related packages. 

This build is failing: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__rosjava_core__ubuntu_xenial_amd64__binary/

The first failing build was trigged by https://github.com/ros/rosdistro/pull/19978 so a revert of that was attempted: https://github.com/ros/rosdistro/pull/20018 but that did not resolve the issue.

Looking into this I believe that the error is related to default behaviors of gradle and what sources are available. I found a few links that look relevant:

https://stackoverflow.com/questions/45781489/error6-0-gradle-dsl-method-not-found-google
https://stackoverflow.com/questions/47152282/gradle-could-not-find-method-google-for-arguments-on-repository-container?noredirect=1&lq=1
https://stackoverflow.com/questions/27470443/could-not-find-method-jcenter-for-arguments-on-repository-container/52039526#52039526

jubeira, 2019-01-23:
See https://github.com/ros/rosdistro/pull/20018; this probably happened because we need the new Gradle version in `rosjava_core`, which hadn't been released.

If the build now passes the reverted new version of `rosjava_build_tools` should work as well.

tfoote, 2019-01-23:
This looks to have worked, There appears to be similar issues with rosjava_extras zeroconf_jmdns_suite

jubeira, 2019-01-30:
Everything solved now; closing the issue.
Thanks @tfoote !

closed 2019-01-30 by jubeira