rosjava/rosjava_core #298: Missing javax.xml.bin for build with Java11

orat, 2019-07-17:
Hi,
rosjava_core can not be build for  java11 because javax.xml.bind is not included in version 11 anymore. 

I do not really understand the build-details of rosjava so I do not know where to add the depencency. There is javax.xml.bind available with version 2.2.11 on jcenter. So it should be easy to add the depencency. 

But in which files?

best regards
Oliver

jubeira, 2019-07-17:
Hi @orat,

Do you have a link with some documentation regarding the bind file? I'm not familiar with the details of Java 11.
Rosjava uses Gradle, perhaps checking out Gradle documentation to build with Java 11 is a good starting point?
If you find a backwards-compatible way of supporting Java 11, feel free to send a PR.

Thanks!

henroth, 2020-03-05:
For anyone else wondering add the following to the dependencies in this file https://github.com/rosjava/rosjava_core/blob/kinetic/apache_xmlrpc_common/build.gradle

compile group: 'javax.xml.bind', name: 'jaxb-api', version: '2.2.4'

Dave-van-der-Meer, 2020-07-23:
> For anyone else wondering add the following to the dependencies in this file https://github.com/rosjava/rosjava_core/blob/kinetic/apache_xmlrpc_common/build.gradle
> 
> compile group: 'javax.xml.bind', name: 'jaxb-api', version: '2.2.4'

I have these dependencies already out of the box and yet I have errors about missing dependencies for javax.xml.bind when running catkin_make.